blheli-configurator/blheli-configurator #8: Gets stuck on "Waiting for data"

sean-duffy, 2016-10-11:
Tried this on OS X El Capitan 10.11.5 with Favourite BLHeli_S ESCs and it gets stuck on the "Waiting for data" screen. Here is the output of the log:

`2016-10-11 @ 13:59:35 -- Running - OS: MacOS, Chrome: 53.0.2785.143, Configurator: 0.0.8
2016-10-11 @ 13:59:48 -- Serial port successfully opened with ID: 3
2016-10-11 @ 13:59:48 -- MultiWii API version received - 1.20.0
2016-10-11 @ 13:59:48 -- Flight controller info, identifier: BTFL, version: 3.0.0
2016-10-11 @ 13:59:48 -- Running firmware released on: Sep 11 2016 00:35:30
2016-10-11 @ 13:59:48 -- Board: AFNA, version: 2
2016-10-11 @ 13:59:48 -- Unique device ID received - 0x66dff484848667267251343`

I'm able to connect fine to the latest BLHeliSuite:
[image]

But this is what I get on the blheli-configurator:
[image]

Is there any way I can get more detailed debug information?

DieHertz, 2016-10-11:
Could you have a look at the developer's console? There may be some messages.

DieHertz, 2016-10-11:
You've checked out the repository from git and trying to run it? Have you executed the build command?
```npm run build```

It's not yet ready for public consumption, I'll add a readme on how to build it etcetera a bit later.

sean-duffy, 2016-10-11:
Ah no I hadn't built it, didn't realise I had to do that. :)

I'm getting "ReferenceError: Unknown plugin "syntax-async-functions" specified in "/Users/SeanDuffy/Programming/projects/blheli-configurator/.babelrc" at 0" now though, but I'll give you a chance to get the readme together.

DieHertz, 2016-10-11:
Sure. Thanks, I'll check what may be wrong with babel plugins, probably the `package.json` definitions.
IIRC, you should also run `npm install` prior to anything else, but I will surely add this to BUILD steps into the readme.
As well as a disclaimer :-)

sean-duffy, 2016-10-11:
Excellent, that did the trick! Reads my setup fine now, great work on this.

closed 2016-10-11 by sean-duffy