blheli-configurator/blheli-configurator #16: Unable to flash Multiple brands of ESCs using remote hex download.

lacymorrow, 2016-10-22:
Even with "ignore improper MCU warnings" enabled I am unable to individually flash each ESC with 
 the latest versions of blheli using the online chooser.

I'm running two Littlebees and two XM20s on motors 1 & 4. All ESC info is read correctly.

 - I flash the first ESS (dys) with the latest dys firmware loaded.
 - I do the same for the littlebees, explicitly choosing the FVT Littlebee 20A. 

 - Expected: Two Escs are flashed with DYS and two with Littlebee, since I explicitly choose the hex before each flash

### Actual

All ESCs are flashed with DYS firmware. The littlebees do not run.

### Solutions

1) Use the official hex file and load them locally, rather than using the chooser. Since I was doing it anyway I went with Multishot, no issues.

2) Use the old ugly, but reliable BLHeli

lacymorrow, 2016-10-22:
[image]

DieHertz, 2016-10-23:
"Ignore inappropriate MCU and Layout" is there to flash firmware different from the original, e.g. DYS XM20A between Graupner Ultra 20A and DYS XM20A, for for flashing ESCs with corrupted settings page.
Btw, you can select MultiShot capable versions in firmware selector.

DieHertz, 2016-10-23:
What exactly is your screenshot demonstrating, just setup?

DieHertz, 2016-10-23:
Thanks, that's a serious bug with caching firmware files.
Ignore inappropriate MCU and Layout actually saves you from problems in that case :)

closed 2016-10-23 by DieHertz

lacymorrow, 2016-10-23:
> Am I right that you are choosing firmware for Littlebee 20A, but another firmware gets flashed? Does this behaviour occur even when flashing only one ESC in a session?
"Ignore inappropriate MCU and Layout" is there to flash firmware different from the original, e.g. DYS XM20A between Graupner Ultra 20A and DYS XM20A, for for flashing ESCs with corrupted settings page.

Yes this is exactly what's happening. The screenshot is after flashing manually with multishot.

> Btw, you can select MultiShot capable versions in firmware selector.

Yes but again, it would flash my littlebees with the firmware chosen for ESC 1 (DYS).

DieHertz, 2016-10-23:
The issue should be fixed in 0.1.3, let me know when you've had a chance to try it out.

lacymorrow, 2016-10-23:
awesome!, i'll try it out just for the hell of it, i'm only running it this way to get rid of my last XM20 ESC

commit 4336265e (2016-10-23): Merge pull request #18 from DieHertz/master. This closes #16

0.1.3