blheli-configurator/blheli-configurator #25: Keyboard arrow keys support on draw bars.

AndersHoglund, 2016-12-20:
A small feature request. When adjusting drawbars in the GUI it is very hard to be precise with a mouse. It would be great if it would be possible to adjust using arrow keys on the keyboard. Or some (+) and (-) buttons on each side of the drawbars, for fine tuning.
For example when selecting tones in BLHeli_S you need to set Beeper Strength values to 91,92,93 and 94. I never managed to select 94, always ended up with 93 or 95, with the mouse. The best would of course have been if tone selection would have been its own parameter with radio buttons. Having certain beeper strength values meaning something else is really some uglu hackish cheating....  
Maybe the GUI could be made with radio buttons anyway, setting fixed selected beeper strength values?

hydralix, 2016-12-29:
I completely agree. Suffice it to + and - to add that adds +1 to the values

DieHertz, 2016-12-29:
I'll do it better, as described in #26. I'll fix sliders not getting focus, that will allow arrow keys. I will also look into the snap algorithm to allow snapping each integer position.

PR #32 (2017-01-03): Fixed slider on-focus behaviour
I wonder why InputRange does not do the same thing out of the box.
Fixes #25, #26
closed 2017-01-03 by DieHertz