blheli-configurator/blheli-configurator #30: DALRC XR20A ESC still timeout with v1.0.3 (related #28)

FPVDrone, 2016-12-27:
[Log - Blheli Configurator App - 20161228.txt](https://github.com/blheli-configurator/blheli-configurator/files/674393/Log.-.Blheli.Configurator.App.-.20161228.txt)

I have tried the v1.0.3 but the same result after "readsetup" without showing anything on the App.  Attached is the log.  Please advise

DieHertz, 2016-12-27:
Try BLHeliSuite if you can. Also try BetaFlight, your issue appeara not related to the configurator, but to CleanFlight.

FPVDrone, 2016-12-27:
Ok Thanks!

DieHertz, 2016-12-27:
@FPVDrone any news on this one?

FPVDrone, 2016-12-28:
I use Mac. Tried to use Wine for using BLHeliSuite but sounded the ESC still timeout.  Not sure why.  I am talking to friend to see whether he has a spare Window notebook for me to test.  Since I am not familiar with BetaFlight, I did not try it.  Meanwhile, I just rewire to make the drone to fly first.  Please close the case.  If I have further finding or update, I shall talk to you again. Thanks!

closed 2016-12-28 by FPVDrone

DieHertz, 2016-12-28:
I'll investigate this as well, probably CF 1.14.x has broken passthrough, maybe it's even caused by my PR cleanflight/cleanflight#2533

FPVDrone, 2016-12-28:
Would like to know if you find anything.  I am using the latest CF 1.14.1   Thanks!

DieHertz, 2016-12-28:
@FPVDrone Oh, 1.14.1. My PR got merged into 1.14.2 only.
I will investigate, though. Soon I'll get my STLink and will be able to debug normally.