blheli-configurator/blheli-configurator #33: Settings are not read back.

AndersHoglund, 2017-01-03:
Noticed that only one ESC was read when clicking "Read Setup" and the settings did not seem correct. At least not Beep and Beacon strenght. Read as "1" when I have the values "80" and "92" resp. BLHeliSuite read them OK. Attaching a logfile.
[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/682206/Log.txt)

AndersHoglund, 2017-01-03:
BLHelisuite log file for reference-
[blhelisuite_Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/682220/blhelisuite_Log.txt)

DieHertz, 2017-01-03:
Thanks for reporting it, @AndersHoglund . By a mistake I have uploaded an incorrect package to the Chrome Web Store.
Please try 1.0.5 and let me know how it works for you when you have time.

AndersHoglund, 2017-01-03:
Thanks. That was a quick fix. All seems to work now, even save settings with a new tone/melody selected via keyboard arrows.

closed 2017-01-03 by AndersHoglund