blheli-configurator/blheli-configurator #39: blheli isnt connecting to esc with new beta flight 3.1

winmorknlz, 2017-01-13:
the blheli conf, isnt getting all my esc when programing, and will sometime recognize all four then other see 3 or even 2 escs?!?!? only while with beta flight 3.1

DieHertz, 2017-01-13:
Provide BLHeli Configurator version number, debug log, name of FC.

winmorknlz, 2017-01-13:
1.07 then 1.0.8

DieHertz, 2017-01-13:
Save debug log after reading setup multiple times and attach it there, please.

winmorknlz, 2017-01-13:
Running - OS: <strong>MacOS</strong>, Chrome: <strong>49.0.2623.112</strong>, Configurator: <strong>1.0.8</strong>
Selecting last used port: /dev/tty.SLAB_USBtoUART
Connecting to: /dev/tty.SLAB_USBtoUART
SERIAL: Connection opened with ID: 154, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 154
MultiWii API version <span style="color: green">received</span> - <strong>1.27.0</strong>
Flight controller info, identifier: <strong>BTFL</strong>, version: <strong>3.1.0</strong>
Running firmware released on: <strong>Jan 11 2017 09:07:22</strong>
Board: <strong>SRF3</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x3600225433571420313737</strong>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 2 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 2
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup <span style="color: green">finished</span>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 2 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 2
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup <span style="color: green">finished</span>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,3 [... cut ...] 69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 2
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,25

DieHertz, 2017-01-13:
Are your certain it does not happen if you flash BF3.0 right now on this setup?
This error generally indicates a wiring or powering issue with the ESC, are the same ESCs numbers having problems or it's completely arbitrary?
I will provide you a test version today, but I've no high expectations that it'll fix it yet.

winmorknlz, 2017-01-13:
yes with beta 3.0 it was fine.. with the newest version it then wouldn't recognize  the escs. and now if i try to flash anything blhel say file isn't avail..

DieHertz, 2017-01-13:
Yeah, that's another issue, thanks for reporting it, will fix in a couple of hours, file names messed up. Could you try it with BF3.0 again just to be sure?

winmorknlz, 2017-01-13:
yes sir!!

winmorknlz, 2017-01-13:
this the log file with the 3.01 beta

Running - OS: <strong>MacOS</strong>, Chrome: <strong>49.0.2623.112</strong>, Configurator: <strong>1.0.8</strong>
Selecting last used port: /dev/cu.SLAB_USBtoUART
Connecting to: /dev/cu.SLAB_USBtoUART
SERIAL: Connection opened with ID: 161, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 161
MultiWii API version <span style="color: green">received</span> - <strong>1.21.0</strong>
Flight controller info, identifier: <strong>BTFL</strong>, version: <strong>3.0.1</strong>
Running firmware released on: <strong>Oct 18 2016 10:35:52</strong>
Board: <strong>SRF3</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x3600225433571420313737</strong>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,3,251,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 1
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 2
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup <span style="color: green">finished</span>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,3,251,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 1
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 2
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup <span style="color: green">finished</span>

DieHertz, 2017-01-13:
Please try these three version in any order and report whether some of them helps with your issue, and which one.

https://dl.dropboxusercontent.com/u/106348265/blheli-configurator-test-delay-100ms.zip
https://dl.dropboxusercontent.com/u/106348265/blheli-configurator-test-delay-250ms.zip
https://dl.dropboxusercontent.com/u/106348265/blheli-configurator-test-delay-500ms.zip

DieHertz, 2017-01-13:
Regarding the issue with "File is unavailable" - it should be solved in about 30 minutes, you may try clearing the cache of your browser. I have temporarily fixed the definitions file to avoid this issue and your Configurator should download the the new file on reconnect or periodically (each 15 minutes if you don't disconnect).

DieHertz, 2017-01-13:
@winmorknlz By the way, your Chrome version is 49, while the latest is 55 something, it may also be the cause of your issue.

winmorknlz, 2017-01-13:
sorry how do i use those link to install?

DieHertz, 2017-01-13:
You unpack one of them to some folder, navigate to `chrome://extensions` in your browser and click `Load unpacked extension…` there, pointing it to the folder where you unpacked archive to. Then a new extension appears in your Chrome installation and you can run it for testing.

winmorknlz, 2017-01-13:
Google chrome now won't allow mac users to update with a the latest version of mos.. 

I'm sorry it won't allow me to use the extension,however now lol…. i flashed the firmware to the 3.01 oldr version to update the escs and the flash will not initiate. This is the only version it will see all 4 esc and worked last night, anyhow it won't flash even in the 3.01any of the escs now, ..

winmorknlz, 2017-01-13:
here is the log file

Running - OS: <strong>MacOS</strong>, Chrome: <strong>49.0.2623.112</strong>, Configurator: <strong>1.0.9</strong>
Selecting last used port: /dev/cu.SLAB_USBtoUART
Connecting to: /dev/cu.SLAB_USBtoUART
SERIAL: Connection opened with ID: 179, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 179
MSP data request timed-out: 1
MSP data request timed-out: 1
MSP data request timed-out: 1
MSP data request timed-out: 1
MSP data request timed-out: 1
MSP data request timed-out: 1
MSP data request timed-out: 1
MSP data request timed-out: 1
MSP data request timed-out: 1
Connection <span style="color: red">failed</span>, ensure that you have selected the <span style="color: yellow">correct port</span> or <span style="color: yellow">restart</span> your device.
MSP data request timed-out: 1
SERIAL: Connection with ID: 179 closed, Sent: 66 bytes, Received: 0 bytes
PortHandler - Removed: /dev/cu.SLAB_USBtoUART,/dev/tty.SLAB_USBtoUART
PortHandler - Found: /dev/cu.SLAB_USBtoUART,/dev/tty.SLAB_USBtoUART
Connecting to: /dev/cu.SLAB_USBtoUART
SERIAL: Connection opened with ID: 180, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 180
MultiWii API version <span style="color: green">received</span> - <strong>1.21.0</strong>
Flight controller info, identifier: <strong>BTFL</strong>, version: <strong>3.0.1</strong>
Running firmware released on: <strong>Oct 18 2016 10:35:52</strong>
Board: <strong>SRF3</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x3600225433571420313737</strong>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,3,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 1
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,8,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 2
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup <span style="color: green">finished</span>

DieHertz, 2017-01-13:
I'm developing the extension on a MacBook with the latest macOS and it allows me to do anything.
What problem your last log is exhibiting? I see that it has correctly read all 4 ESCs.
If you can't try the 3 versions provided via ZIP archives - we won't be able to check whether it fixed your issue :(

winmorknlz, 2017-01-13:
blheli isn't flashing anything.. it presses the flash button but nothing happens on either 3.0 or 3.1;/

DieHertz, 2017-01-13:
That's my fault! I have commented this line, lol :-) But you still have to figure out how to install test revision.

winmorknlz, 2017-01-13:
well for not till we can figure out the test rev. if i could flash all my escs with the same revision in either beta flight we will work around the newest beta flight..thank you for acting so quick.. these quads are so fickle!! lol but we love thm!! lol

DieHertz, 2017-01-13:
Pushed 1.0.10 to store which fixes the `Flash` button doing nothing :) Wait up to 30 minutes for it to become available.

I have also updated the three files attached above to include this fix as well, in case you're going to try it again I'm duplicating these links:
https://dl.dropboxusercontent.com/u/106348265/blheli-configurator-test-delay-100ms.zip
https://dl.dropboxusercontent.com/u/106348265/blheli-configurator-test-delay-250ms.zip
https://dl.dropboxusercontent.com/u/106348265/blheli-configurator-test-delay-500ms.zip

winmorknlz, 2017-01-13:
thanks, i will tinker with them to see if i can get them to work in this chrome app.. i may have to go buy a windows computer ;( ugh chrome said that certain apple version will no longer receive  updates. thus making them eventually obsolete.. thank you again!!

winmorknlz, 2017-01-13:
one more log  3.1 beta flight all but esc2 working won't flash esc2

Running - OS: <strong>MacOS</strong>, Chrome: <strong>49.0.2623.112</strong>, Configurator: <strong>1.0.10</strong>
Last used port wasn't saved "yet", auto-select disabled.
Connecting to: /dev/cu.Bluetooth-Modem
Failed to connect to the port.
SERIAL: Failed to open serial port
Failed to open serial port
<span style="color: red">Failed</span> to open serial port
Connecting to: /dev/cu.SLAB_USBtoUART
SERIAL: Connection opened with ID: 4, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 4
MultiWii API version <span style="color: green">received</span> - <strong>1.27.0</strong>
Flight controller info, identifier: <strong>BTFL</strong>, version: <strong>3.1.0</strong>
Running firmware released on: <strong>Jan 11 2017 09:07:22</strong>
Board: <strong>SRF3</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x3600225433571420313737</strong>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 2 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,8,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 2
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup <span style="color: green">finished</span>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 2 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,8,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 2
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup <span style="color: green">finished</span>
Reading setup
[... 389 lines cut ...]
sending cmd_DeviceRead 800 128
received cmd_DeviceRead ACK_OK 800 250,235,148,0,251,195,232,19,195,19,248,234,40,250,235,52,0,251,138,71,139,72,195,235,148,2,64,2,194,44,120,2,235,196,254,123,0,234,196,84,15,78,250,195,234,152,250,64,5, [... cut ...]
sending cmd_DeviceRead d80 128
received cmd_DeviceRead ACK_OK d80 34,117,101,0,117,102,0,117,151,222,117,151,173,117,129,192,67,255,128,117,239,6,117,169,0,113,113,117,128,255,117,241,240,117,164,0,117,128,255,117,212,255,117,144,0,117,242,63,117,165,63,117,144,0,117,213,63,117,166,1,117,227,64,117,226,2,113,113,228,248,246,216,253,113,129,18,18,65,120,152,134,98,117,48,1,194,175,18,5,144,18,5,159,18,5,136,18,5,165,18,5,136,18,5,171,18,5,136,177,105,194,175,117,101,0,117,102,0,120,250,121,250,48,133,7,217,251,216
s

winmorknlz, 2017-01-13:
this is after flashing 3.0 beta back, and the flash worked great, now to ref lash beta 3.1 with the latest esc firmware

Running - OS: <strong>MacOS</strong>, Chrome: <strong>49.0.2623.112</strong>, Configurator: <strong>1.0.10</strong>
Selecting last used port: /dev/cu.SLAB_USBtoUART
Connecting to: /dev/cu.SLAB_USBtoUART
SERIAL: Connection opened with ID: 8, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 8
MultiWii API version <span style="color: green">received</span> - <strong>1.21.0</strong>
Flight controller info, identifier: <strong>BTFL</strong>, version: <strong>3.0.1</strong>
Running firmware released on: <strong>Oct 18 2016 10:35:52</strong>
Board: <strong>SRF3</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x3600225433571420313737</strong>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,3,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 1
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,8,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 2
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,6,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,4,255,255,255,7,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup <span style="color: green">finished</span>
ESC 2: flashing <strong>started</strong>
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 178,232,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,3,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DevicePageErase 0 13
received cmd_DevicePageErase ACK_OK 1a00 13
sending cmd_DeviceWrite 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,3,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
received cmd_DeviceWrite ACK_OK 1a00 0
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,4,255,255,255,3,255,40,80,4,255,3,255,122,255,1,1,255,255,1,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
sending cmd_DeviceRead 0 3
received cmd_DeviceRead ACK_OK 0 2,25,253
sending cmd_DevicePageErase 0 1
received cmd_DevicePageErase ACK_OK 200 1
sending cmd_DeviceWrite 200 2,28,0
received cmd_DeviceWrite ACK_OK 200 0
sending cmd_DeviceRead 200 3
received cmd_DeviceRead ACK_OK 200 2,28,0
sending cmd_DevicePageErase 0 0
received cmd_DevicePageErase ACK_OK 0 0
sending cmd_DeviceRead 0 128
received cmd_DeviceRead ACK_OK 0 255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255 [... cut ...] 5,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
sending cmd_DeviceRead 1b00 128
received cmd_DeviceRead ACK_OK 1b00 255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,25

DieHertz, 2017-01-14:
Only logs from modified packages with delay can be helpful now, I see the same picture in all logs till now - read from ESC fails. If that isn't solved by adding a delay - it may be an issue in BF3.1 which we don't know about yet.

@winmorknlz you couldn't install the modified versions?

winmorknlz, 2017-01-15:
Actually it did great with 3.0 not 3.1. So I think it bf3.1

Sent from my U.S. Cellular® Smartphone

  to esc with new beta flight 3.1 (#39)

glambert22, 2017-07-15:
I see this issue is still open. I am having the same issues. Using BF 3.1.1. FC: SP Racing F3. Emax 12A BLheli. 

[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1150605/Log.txt)

DieHertz, 2017-07-15:
Your ESCs don't have a bootloader.

HighTechPauper, 2017-12-20:
I found that having port peripherals enabled messed up the comm with esc's. Disabled port peripherals (RunCam, SmartAudio, Tramp) and all was well again. So now I just remember to disable the ports mods before I do any esc work

closed 2020-11-13 by mikeller