blheli-configurator/blheli-configurator #42: Flash keeps failing

r1b4z01d, 2017-01-16:
Running - OS: <strong>MacOS</strong>, Chrome: <strong>55.0.2883.95</strong>, Configurator: <strong>1.0.10</strong>
Connecting to: /dev/cu.Bluetooth-Incoming-Port
SERIAL: Connection opened with ID: 21, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 21
MSP data request timed-out: 1
MSP data request timed-out: 1
SERIAL: Connection with ID: 21 closed, Sent: 18 bytes, Received: 0 bytes
PortHandler - Found: /dev/cu.SLAB_USBtoUART,/dev/tty.SLAB_USBtoUART
Connecting to: /dev/cu.SLAB_USBtoUART
SERIAL: Connection opened with ID: 22, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 22
MultiWii API version <span style="color: green">received</span> - <strong>1.21.0</strong>
Flight controller info, identifier: <strong>BTFL</strong>, version: <strong>3.0.1</strong>
Running firmware released on: <strong>Oct 18 2016 10:29:13</strong>
Board: <strong>RMDO</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x2400175533571220303834</strong>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 7,147,97,2
sending cmd_DeviceReadEEprom 0 112
received cmd_DeviceReadEEprom ACK_OK 0 14,6,21,9,9,4,255,3,255,10,3,2,1,85,170,1,255,255,255,255,255,3,255,255,255,15,223,40,80,4,255,2,0,122,255,0,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,83,78,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,65,109,56,65,35,32,32,32,42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 7,147,97,2
sending cmd_DeviceReadEEprom 0 112
received cmd_DeviceReadEEprom ACK_OK 0 14,6,21,9,9,4,255,3,255,10,3,1,1,85,170,1,255,255,255,255,255,3,255,255,255,24,227,40,80,4,255,2,0,122,255,0,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,83,78,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,65,109,56,65,35,32,32,32,42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 7,147,97,2
sending cmd_DeviceReadEEprom 0 112
received cmd_DeviceReadEEprom ACK_OK 0 14,6,21,9,9,4,255,3,255,10,3,1,1,85,170,1,255,255,255,255,255,3,255,255,255,36,231,40,80,4,255,2,0,122,255,0,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,83,78,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,65,109,56,65,35,32,32,32,42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 7,147,97,2
sending cmd_DeviceReadEEprom 0 112
received cmd_DeviceReadEEprom ACK_OK 0 14,6,21,9,9,4,255,3,255,10,3,2,1,85,170,1,255,255,255,255,255,3,255,255,255,30,236,40,80,4,255,2,0,122,255,0,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,83,78,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,65,109,56,65,35,32,32,32,42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
Reading setup <span style="color: green">finished</span>
ESC 1: flashing <strong>started</strong>
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 7,147,97,2
sending cmd_DeviceReadEEprom 0 112
received cmd_DeviceReadEEprom ACK_OK 0 14,6,21,9,9,4,255,3,255,10,3,2,1,85,170,1,255,255,255,255,255,3,255,255,255,15,223,40,80,4,255,2,0,122,255,0,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,83,78,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,65,109,56,65,35,32,32,32,42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
sending cmd_DeviceWriteEEprom 60 42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
received cmd_DeviceWriteEEprom ACK_OK 60 0
sending cmd_DeviceReadEEprom 60 16
received cmd_DeviceReadEEprom ACK_OK 60 42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
sending cmd_DeviceWrite 40 223,205,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
received cmd_DeviceWrite ACK_OK 40 0
sending cmd_DeviceRead 40 2
received cmd_DeviceRead ACK_D_GENERAL_ERROR 40 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 40 0
ESC 1: flashing <span style="color: red">failed</span> - <strong>Error: cmd_DeviceRead ACK_D_GENERAL_ERROR
    at recvCallback (chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/_4way_if.js:198:37)
    at chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/_4way_if.js:296:29
    at Array.forEach (native)
    at Object.onread (chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/_4way_if.js:278:18)
    at read_serial (chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/serial_backend.js:234:15)
    at EventImpl.dispatchToListener (extensions::event_bindings:388:22)
    at Event.publicClassPrototype.(anonymous function) [as dispatchToListener] (extensions::utils:149:26)
    at EventImpl.dispatch_ (extensions::event_bindings:372:35)
    at dispatchArgs (extensions::event_bindings:244:26)
    at dispatchEvent (extensions::event_bindings:253:7)</strong>
ESC 2: flashing <strong>started</strong>
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 7,147,97,2
sending cmd_DeviceReadEEprom 0 112
received cmd_DeviceReadEEprom ACK_OK 0 14,6,21,9,9,4,255,3,255,10,3,1,1,85,170,1,255,255,255,255,255,3,255,255,255,24,227,40,80,4,255,2,0,122,255,0,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,83,78,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,65,109,56,65,35,32,32,32,42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
sending cmd_DeviceWriteEEprom 60 42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
received cmd_DeviceWriteEEprom ACK_OK 60 0
sending cmd_DeviceReadEEprom 60 16
received cmd_DeviceReadEEprom ACK_OK 60 42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
sending cmd_DeviceWrite 40 223,205,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
received cmd_DeviceWrite ACK_OK 40 0
[... 87 lines cut ...]
ESC 1: flashing <strong>started</strong>
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 7,147,97,2
sending cmd_DeviceReadEEprom 0 112
received cmd_DeviceReadEEprom ACK_OK 0 14,6,21,9,9,4,255,3,255,10,3,2,1,85,170,1,255,255,255,255,255,3,255,255,255,15,223,40,80,4,255,2,0,122,255,0,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,83,78,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,65,109,56,65,35,32,32,32,42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
sending cmd_DeviceWriteEEprom 60 42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
received cmd_DeviceWriteEEprom ACK_OK 60 0
sending cmd_DeviceReadEEprom 60 16
received cmd_DeviceReadEEprom ACK_OK 60 42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
sending cmd_DeviceWrite 40 223,205,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
received cmd_DeviceWrite ACK_OK 40 0
sending cmd_DeviceRead 40 2
received cmd_DeviceRead ACK_D_GENERAL_ERROR 40 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 40 0
ESC 1: flashing <span style="color: red">failed</span> - <strong>Error: cmd_DeviceRead ACK_D_GENERAL_ERROR
    at recvCallback (chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/_4way_if.js:198:37)
    at chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/_4way_if.js:296:29
    at Array.forEach (native)
    at Object.onread (chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/_4way_if.js:278:18)
    at read_serial (chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/serial_backend.js:234:15)
    at EventImpl.dispatchToListener (extensions::event_bindings:388:22)
    at Event.publicClassPrototype.(anonymous function) [as dispatchToListener] (extensions::utils:149:26)
    at EventImpl.dispatch_ (extensions::event_bindings:372:35)
    at dispatchArgs (extensions::event_bindings:244:26)
    at dispatchEvent (extensions::event_bindings:253:7)</strong>

DieHertz, 2017-01-16:
Weird, fails very consistently at the first flash read operation, which leads me to believe your Atmel ESC may have fuses set to disallow readback, no way to alter them via bootloader.
Could you try flashing it with BLHeliSuite and save the log in there?
I could also provide you a test revision with some changes to Atmel flashing procedure and a delay between each operation, but if it's really the fuses - you'll have to use Atmel ICSP flashing via avrdude.

Nitrokillah, 2017-04-29:
Oh man this is too bad ... I thought the Hole point of using these Chrome app Was to get away from using Atmel ICSP to FLASH YOUR ESCs .  
I've had a similar issue  using  Betaflight F3 FC  & DYS XSD30A V2 .  
Every time i go to Flash my ESCs on The Chrome App ( Mac )  one of the ESC Seems to other died / diapers form the List in the app .  first time was ESC 1  , swapped the ESC with a new one ,  flashed them and now it's ESC 3 that is Gone  ...  

Running - OS: <strong>MacOS</strong>, Chrome: <strong>57.0.2987.133</strong>, Configurator: <strong>1.0.11</strong>
PortHandler - Found: /dev/tty.usbmodem1421,/dev/cu.usbmodem1421
Connecting to: /dev/tty.usbmodem1421
SERIAL: Connection opened with ID: 11, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 11
MultiWii API version <span style="color: green">received</span> - <strong>1.31.0</strong>
Flight controller info, identifier: <strong>BTFL</strong>, version: <strong>3.1.7</strong>
Running firmware released on: <strong>Apr  3 2017 21:43:25</strong>
Board: <strong>BFF3</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x2000354234571520303137</strong>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,3,255,255,255,4,252,78,152,3,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,3,255,255,255,4,252,78,152,3,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 1
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,100,1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,100,1
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,3,255,255,255,4,252,78,152,3,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup <span style="color: green">finished</span>
sending cmd_InterfaceExit 0 0
received cmd_InterfaceExit ACK_OK 0 0
SERIAL: Connection with ID: 11 closed, Sent: 594 bytes, Received: 1066 bytes
PortHandler - Removed: /dev/tty.usbmodem1421,/dev/cu.usbmodem1421
PortHandler - Found: /dev/tty.usbmodem1421,/dev/cu.usbmodem1421
Connecting to: /dev/tty.usbmodem1421
SERIAL: Connection opened with ID: 12, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 12
MultiWii API version <span style="color: green">received</span> - <strong>1.31.0</strong>
Flight controller info, identifier: <strong>BTFL</strong>, version: <strong>3.1.7</strong>
Running firmware released on: <strong>Apr  3 2017 21:43:25</strong>
Board: <strong>BFF3</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x2000354234571520303137</strong>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,3,255,255,255,4,252,78,152,3,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,3 [... cut ...]
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,3,255,255,255,4,252,78,152,3,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup <span style="color: green">finished</span>

Please HELP ...

HighTechPauper, 2017-12-20:
I found that having port peripherals enabled messed up the comm with esc's. Disabled port peripherals (RunCam, SmartAudio, Tramp) and all was well again. So now I just remember to disable the ports mods before I do any esc work

Nitrokillah, 2017-12-20:
I got an Arduino and did NOT manage to flash them :( 
DYS  Customer service was really REALLY Bad 👎  
I had to send them to the the Shop , the Shop sent it do the Distribution, and the Distribution Sent it to DYS ... ( some longness that DYS just did not want to understand  )  
After about 3 months I got the Same exact ESCs back and they don't even work.  
(GOOD job DYS )     
 I've Moved on since then . got some 
Raceflight esc that were noisy  
AIKON-V3-SEFM Cristal clear 
Betaflight 32Bit 35A  witch are a bit noisy . 

all In all I just want to say that due to this issue I am too Scared to flash ESCs because of the Hassle I had to got thru with DYS and BLheli not working on a MAC . ( I only Do esc setups on a PC but I don't flash them anymore it cost me too much )
I am not the only one who has had issues with this ; Kevin Dougherty "StingersSwarm" has had the problems .

closed 2020-11-13 by mikeller