blheli-configurator/blheli-configurator #44: Can not read Emax BLheli ESC's

silentwindblows, 2017-02-06:
I just installed blheli in chrome, went back into Cleanflight configurator to disable UART 3 MSP, which is normally used for CC3D boards flex port. Then i successfully connected to the board in bLheli chrome, but every time i click read setup, the log window just shows "reading setup", then "read setup finished". Nothing is displayed after this. All VCP Drivers are installed, latest cleanflight firmware on the CC3D flight Controller board, so why isn't BLheli chrome not displaying the ESC settings after reading reading setup, blank interface, just a little note top left of screen regarding moving propellers and power the ESC's. Very strange. I've tried all 3 ports on the CC3D (FTDI USB UART adaptor on flex & main port). Green light is solid when i connect to CC3D within BLheli chrome, so obviously the CC3D enters bootloader mode, but it isnt showing ESC settings after reading

DieHertz, 2017-02-06:
Nothing I can say without a debug log.

Ludwant, 2017-05-21:
[BLHeli Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1016603/BLHeli.Log.txt)
Hi, I am having the same problem with my Naze32 and Emax ESCs. It can connect, and when I hit Read Setup it makes the ESCs beep, but nothing shows up in the configurator.

DieHertz, 2017-05-21:
Your ESCs don't have an appropriate bootloader or are otherwise not suited for passthrough

closed 2017-05-21 by DieHertz