blheli-configurator/blheli-configurator #45: Cannot read ESC's

jorimax, 2017-02-13:
I am trying to connect to my esc's via the BLheli suite but I keep on getting errors. (I have done this before so I know how it works) The tool finds ESC's in Silabs mode and I have used option E
When I connect the battery I hear the correct tones. Then I select "Read Setup" and a pop-up shows rebooting the FC and I get the error connections to fc failed check port and power etc.. 
I can connect fine with Betaflight. I have flashed the laters Spracingf3

2017-02-13 @ 23:45:58 -- Serial port successfully opened with ID: 3
2017-02-13 @ 23:45:58 -- MultiWii API version received - 1.31.0
2017-02-13 @ 23:45:58 -- Flight controller info, identifier: BTFL, version: 3.1.5
2017-02-13 @ 23:45:58 -- Running firmware released on: Feb 7 2017 22:30:44
2017-02-13 @ 23:45:58 -- Board: SRF3, version: 0
2017-02-13 @ 23:45:58 -- Unique device ID received - 0x3e002b5633571520313639
2017-02-14 @ 00:00:39 -- Serial port successfully closed

Could there be something wrong with the FC or USB port?
Thanks a lot!
Jori

[image]

[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/772641/Log.txt)

andymac4182, 2017-02-13:
Hi @jorimax,
This repo is for the chrome app. https://chrome.google.com/webstore/detail/blheli-configurator/mejfjggmbnocnfibbibmoogocnjbcjnk

Have you given that a go?

Cheers,
Andrew

jorimax, 2017-02-13:
> :
> Hi @jorimax <https://github.com/jorimax>,
> This repo is for the chrome app. https://chrome.google.com/
> webstore/detail/blheli-configurator/mejfjggmbnocnfibbibmoogocnjbcjnk
>
> Have you given that a go?
>
> Cheers,
> Andrew

yes i have and the same result.
thx a lot
joep

jorimax, 2017-02-13:
the log is actually from the chrome app.

Hi @jorimax <https://github.com/jorimax>,
This repo is for the chrome app. https://chrome.google.com/
webstore/detail/blheli-configurator/mejfjggmbnocnfibbibmoogocnjbcjnk

Have you given that a go?

Cheers,
Andrew

andymac4182, 2017-02-14:
Just confirming you have the ESCs powered up from the battery when you are attempting to connect?

DieHertz, 2017-02-14:
Looks like something wrong with passthrough or serial connection, I'll check my SPRF3 on BF3.1.5 tomorrow.

jorimax, 2017-02-14:
yes i do

jorimax, 2017-02-14:
thanks a lot!

jorimax, 2017-02-14:
Hi Andrew,

Thanks a lot for your time!

I forgot to mention that the quad is actually flying, I managed to configure the ESC's after the build.
One thing I noticed is that one ground wire from a ESC is "almost" touching the outer case of the USB port. Could this be the problem?

The strange thing is that I can connect just fine and use BetaFlight etc... so the USB port should be ok. But when I connect the battery it goes wrong. 

Thanks a lot for your help!

Kr,
Joep

jorimax, 2017-02-15:
Hi Andrey,

Did you managed to test it?
It seems indeed an issue with the passthrough.
I also noticed that BF stops responding once I connect a battery?

thx,
Joep

DieHertz, 2017-02-15:
Checked SPRF3 on 3.1.5, passthrough and BFC work for me.
Your issue looks very similar to what happens when USB UART is shared with some other function as OSD or SPORT telemetry. Any chance you have something connected to UART1?

jorimax, 2017-02-15:
UART 1 is USB I disabled this one and as a result I needed to reflash.
I use UART 2 for 5v for a LED strip so this might be the issue???
UART 3 is Serial RX

DieHertz, 2017-02-15:
MSP should be enabled on UART1, and nothing should be connected to it, otherwise it interferes with normal USB operation, that usually happens when you apply power (e.g. connect your battery).

Do you have a logic analyzer? You could hook it up to UART1 and see what happens when you connect the battery.

jorimax, 2017-02-15:
I have MSP on uart 1 and 2
But I think I have indeed connected either the led or the OSD to uart 1
So I need to move the OSD to uart 2 and do not attach anything to uart 1?

thx!

DieHertz, 2017-02-15:
Yes, precisely! Move the OSD and leave UART1 alone, that should fix your issue.

jorimax, 2017-02-15:
man this took me hours!
Thanks a lot!
really appreciated!

I let you know if it's fixed!

jorimax, 2017-02-15:
Can I take only 5V from uart1?

DieHertz, 2017-02-15:
Sure, all 5V outputs are equal and internally connected.

jorimax, 2017-02-15:
Problem solved!
So happy!

Thanks a lot!

DieHertz, 2017-02-15:
Glad you made it.
Apologies for not pin-pointing the cause earlier. I suggest you join BetaFlight Slack Channel via betaflight.nz, there are lots of knowledgeable people and BF devs.

closed 2017-02-15 by DieHertz

jorimax, 2017-02-16:
I will thanks again!