blheli-configurator/blheli-configurator #46: Only 3 motors show up in BlHeli?

dkeebler, 2017-02-26:
Any ideas why only motors 123 show up. I checked solders not sure what else it might be. Thanks.

r1b4z01d, 2017-02-26:
@dkeebler try using motor pin 5 on the esc that is not working. Does it recognize all 4 this way?

dkeebler, 2017-02-26:
I have the Crazepony BLHeli-S 4-in-1 10A Mini ESC only 4 motors.

r1b4z01d, 2017-02-26:
Let me rephrase that. Try putting the signal wire from the ESC #4 to the esc pad number 5 on the flight controller.

dkeebler, 2017-02-28:
It's a micro with only 4 esc pads. I've ordered a new fc and esc's. thanks.

Apex808, 2017-06-03:
i have same problem cant find any fixes? copter flys so everything works just cant flash #4 because it does not show up?? anyone help thanks
[HobbyWigXrotor30AescLog.txt](https://github.com/blheli-configurator/blheli-configurator/files/1049641/HobbyWigXrotor30AescLog.txt) my log

dkeebler, 2017-06-03:
I ended up returning mine and purchasing new ones with no issues.

dragonFPV, 2017-09-16:
when you say you bought a new one, you mean fc?

dkeebler, 2017-09-16:
No I ordered another micro esc all in one.

johnnephin, 2017-10-02:
Was able to flash only 3 ESC's with 4 showing up.  After closing the application only the 3 that flashed successfully show up.  After hitting read several times the missing ESC showed and let me flash it but the 2 others disappeared.

HighTechPauper, 2017-12-20:
I found that having port peripherals enabled messed up the comm with esc's. Disabled port peripherals (RunCam, SmartAudio, Tramp) and all was well again. So now I just remember to disable the ports mods before I do any esc work

closed 2020-11-13 by mikeller