blheli-configurator/blheli-configurator #47: crazy spinn and restart on ms25a

Gonefpv, 2017-02-28:
I remove cap for ms25a racestart esc and upgrade software to bl heli s 16.6. 

I tested on MS , d300 and d600 always can normal armed and slowly flying is ok but hard landing or shot little in arm result in restart of esc's. I change esc on other xm20a with MS and all working good so its something with this software on MS25A, i tested even 16.53 tones and the same (even worst)

DieHertz, 2017-02-28:
I once had restarts on landing and hard manoeuvres, adding a big low ESR power capacitor solved it.

Gonefpv, 2017-02-28:
what value of capacitor works on your setup?

DieHertz, 2017-02-28:
The ESCs are really bad, so I used 1000uF 35V caps, you could try 470uF

Gonefpv, 2017-03-01:
I try both 470uF and 1000uF all the same ....

closed 2020-11-13 by mikeller