blheli-configurator/blheli-configurator #49: PPM Center Throttle slider wont accept certain values in bidirectional mode

Zappelphilipp, 2017-04-17:
setting the ESC's to Bidirectional (or Bidirectional Reversed) sets the PPM Center Throttle value to 1488. By clicking on the slider (to set it to 1500 for example), it jumps right to 1508 which seems to be the minimum even though it was 1488 by default.
screenshot:
http://imgur.com/a/6Xb03

ignore the error of ESC #1, this happens on all ESC's i tried. (RS35A on 16.65)

bdelapen, 2017-05-22:
I can verify that this is indeed the case. I have the exact same issue running on a Mac. My current workaround is to complete the setup using the Chrome Extension, then boot a Windows VM so that I can set the center throttle value to 1500 on the ESCs. More and more I have been working with 3D quads so it would be great if this could be fixed. I have tried multiple different flight controllers and at least 5 different ESC firmwares, BLHeli 16.6.

bdelapen, 2017-05-22:
Hi @DieHertz,
OK, I found the bug, however I know nothing about how to correct it and get an updated version. I could probably read up on doing it locally for myself but that wouldn't help other people.

File:           blheli-configurator/js/blheli_settings_description.js
Line:           578
Current value:
                name: 'PPM_CENTER_THROTTLE', type: 'number', min: **_1504_**, max: 2020, step: 1, label: 'escPPMCenterThrottle',
Should be:
                name: 'PPM_CENTER_THROTTLE', type: 'number', min: **_1000_**, max: 2020, step: 1, label: 'escPPMCenterThrottle',

Thanks for taking a minute to look at this!

- Birger

DieHertz, 2017-05-23:
Thanks for looking into it, @bdelapen, I'll release a fix today.

closed 2017-05-23 by DieHertz

bdelapen, 2017-05-23:
Thanks Andrey, you Rock!!!

Being a Mac person It was always a chore to work on and flash ESCs, your work has made it a pleasure.

- Birger

PR #56 (2017-05-23): Fix PPM Center Throttle Slider
Fixes #49