blheli-configurator/blheli-configurator #51: ESC ( DYS XSD30A V2 )  Disappear after flashing on Mac

Nitrokillah, 2017-04-30:
Oh man this is too bad ... I thought the Hole point of using these Chrome app Was to get away from using Atmel ICSP to FLASH  ESCs .
I am using Betaflight F3 FC & DYS XSD30A V2 .
 i  Flashed my ESCs on The Chrome App ( Mac ) one of the ESC Seems to have " "died " " or has diapers form the List. 
First time was ESC 1 .  So i swapped the ESC with a new one . 
I flashed all to have the Same FW on all of them .   
# Now the  ESC 3 that is Gone ...

Running - OS: MacOS, Chrome: 57.0.2987.133, Configurator: 1.0.11
PortHandler - Found: /dev/tty.usbmodem1421,/dev/cu.usbmodem1421
Connecting to: /dev/tty.usbmodem1421
SERIAL: Connection opened with ID: 11, Baud: 115200
Serial port successfully opened with ID: 11
MultiWii API version received - 1.31.0
Flight controller info, identifier: BTFL, version: 3.1.7
Running firmware released on: Apr 3 2017 21:43:25
Board: BFF3, version: 0
Unique device ID received - 0x2000354234571520303137
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,3,255,255,255,4,252,78,152,3,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,3,255,255,255,4,252,78,152,3,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 1
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,100,1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,100,1
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,3,255,255,255,4,252,78,152,3,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup finished
sending cmd_InterfaceExit 0 0
received cmd_InterfaceExit ACK_OK 0 0
SERIAL: Connection with ID: 11 closed, Sent: 594 bytes, Received: 1066 bytes
PortHandler - Removed: /dev/tty.usbmodem1421,/dev/cu.usbmodem1421
PortHandler - Found: /dev/tty.usbmodem1421,/dev/cu.usbmodem1421
Connecting to: /dev/tty.usbmodem1421
SERIAL: Connection opened with ID: 12, Baud: 115200
Serial port successfully opened with ID: 12
MultiWii API version received - 1.31.0
Flight controller info, identifier: BTFL, version: 3.1.7
Running firmware released on: Apr 3 2017 21:43:25
Board: BFF3, version: 0
Unique device ID received - 0x2000354234571520303137
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
Reading setup finished
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,3,255,255,255,4,252,78,152,3,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,3,255,255,255,4,252,78,152,3,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 1
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,100,1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,100,1
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,5,33,255,255,255,255,255,255,9,255,1,255,85,170,1,255,255,255,255,255,3,255,255,255,4,252,78,152,3,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,51,48,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup finished

I checked the LED is still lit on the ESC not showing up in the list  ... 
Please HELP ... 
tell me these ESCs aren't dead already  ...
It also seems like there are a few people that have had similar issue on different ESCs...

DieHertz, 2017-04-30:
You don't have to use Atmel ICSP, you have to use an Arduino and BLHeliSuite to flash using C2-interface.
Logs after the flash failure are useless, what's needed are logs from the failed flash attempt.

HighTechPauper, 2017-12-20:
I found that having port peripherals enabled messed up the comm with esc's. Disabled port peripherals (RunCam, SmartAudio, Tramp) and all was well again. So now I just remember to disable the ports mods before I do any esc work

closed 2020-11-13 by mikeller