blheli-configurator/blheli-configurator #55: Possible branch of BLHeli configurator for 32-bit ESCs

aagm85, 2017-05-22:
Hi,
I am a control engineer from Brisbane in Australia and have been developing a multirotor ESC that uses an STM32 processor. Hardware prototypes have been produced and currently I am working on customised firmware that aims to make the most of the 72 MHz processor. I am interested to know if there is any interest for developing a branch of BLHeli configurator that works with this ESC. Considering the popularity of BLHeli configurator, I thought this might be a good idea. Also, BLHeli_32 is closed-source and will not support BLHeli Configurator, so this would be an opportunity to extend it to 32-bit ESCs. The plan would be to for me incorporate the same UART commands as used by BLHeli in the ESC firmware, such that the configurator would look and work essentially the same. Anyway, I'm interested to hear thoughts.
Cheers,
Alex.

DieHertz, 2017-05-23:
We may just add some tab interface to allow support for different ESCs. I also work on firmware for existing 32-bit ESCs as well as BLHeli32 ESCs, they will use a different

DieHertz, 2017-05-23:
We may just add some tab interface to allow support for different ESCs. I also work on firmware for existing 32-bit ESCs as well as BLHeli32 ESCs, I will likely use a different protocol though.
BLHeli32 support may also be there if I have enough time to reverse-engineer it.

aagm85, 2017-05-23:
Thanks for your response. Perhaps you might be interested in working together on a protocol? If not, my protocol will be well documented to ensure smooth integration. I would like to help contribute I some way, although it will take me a bit to get up to speed with your code (my skills are in C and assembly).

closed 2020-11-13 by mikeller