blheli-configurator/blheli-configurator #57: Flash keeps failing for blheli 15 amp escs

dubbadawg, 2017-05-29:
first time drone build from a kit.
Came with 4 15a escs with no markings.
Claimed to be blheli.
trying to flash them and configure/calibrate them, but no luck so far.

Running - OS: <strong>MacOS</strong>, Chrome: <strong>58.0.3029.110</strong>, Configurator: <strong>1.0.14</strong>
Running - OS: <strong>MacOS</strong>, Chrome: <strong>58.0.3029.110</strong>, Configurator: <strong>1.0.13</strong>
Running - OS: <strong>MacOS</strong>, Chrome: <strong>58.0.3029.110</strong>, Configurator: <strong>1.0.13</strong>
Running - OS: <strong>MacOS</strong>, Chrome: <strong>58.0.3029.110</strong>, Configurator: <strong>1.0.13</strong>
Running - OS: <strong>MacOS</strong>, Chrome: <strong>58.0.3029.110</strong>, Configurator: <strong>1.0.13</strong>
Last used port wasn't saved "yet", auto-select disabled.
Uncaught Error: Not found chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/libraries/q.js 134 29 Error: Not found
    at Object.callback (chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/flashing_helpers.js:170:29)
    at safeCallbackApply (extensions::sendRequest:21:15)
    at handleResponse (extensions::sendRequest:74:7)
Connecting to: /dev/tty.SLAB_USBtoUART
SERIAL: Connection opened with ID: 15, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 15
MultiWii API version <span style="color: green">received</span> - <strong>1.31.0</strong>
Flight controller info, identifier: <strong>BTFL</strong>, version: <strong>3.1.7</strong>
Running firmware released on: <strong>Apr  3 2017 22:12:24</strong>
Board: <strong>AFNA</strong>, version: <strong>2</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x670ff575751897267183706</strong>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
[... 334 lines cut ...]
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>

DieHertz, 2017-05-29:
Downgrade to BF 3.0.1 to configure your ESCs, then go back to BF 3.1.7.
BF 3.1.7 has some issues with passthrough.

Which FC your kit has? It says Naze32 target, but is it really a F1?

DieHertz, 2017-05-30:
By the way, do these motors rotate when you try it from the motor tab? There's still a possibility ESCs are not connected to FC motor outputs.

HighTechPauper, 2017-12-20:
I also found that having port peripherals enabled messed up the comm with esc's. Disabled port peripherals (RunCam, SmartAudio, Tramp) and all was well again. So now I just remember to disable the ports mods before I do any esc work

timelord666, 2018-04-28:
how disable port mods?

closed 2020-11-13 by mikeller