blheli-configurator/blheli-configurator #62: ESCs disappeared

Wired99, 2017-07-06:
I tried flashing the 16.6 fw onto a Eachine OmnibusF3 FC with Betaflight.
Flashing failed but I also am not able to see my ESC anymore.
BLHeli PT connects find, reads ok but shows an empty
I hope this log file helps

[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1127770/Log.txt)
 screen.

dragonFPV, 2017-09-16:
I have the same problem, if you find any solution and you can wrtie me, I will be very thankful

HighTechPauper, 2017-12-20:
I found that having port peripherals enabled messed up the comm with esc's. Disabled port peripherals (RunCam, SmartAudio, Tramp) and all was well again. So now I just remember to disable the ports mods before I do any esc work

closed 2020-11-13 by mikeller