blheli-configurator/blheli-configurator #65: Main configuration screen not loading

farrukh2102, 2017-07-21:
Running - OS: Windows, Chrome: 59.0.3071.115, Configurator: 1.0.14
2017-07-21 @ 18:35:46 -- Serial port successfully opened with ID: 3
2017-07-21 @ 18:35:46 -- MultiWii API version received - 1.35.0
2017-07-21 @ 18:35:46 -- Flight controller info, identifier: CLFL, version: 2.0.5
2017-07-21 @ 18:35:46 -- Running firmware released on: Jul 9 2017 15:03:36
2017-07-21 @ 18:35:46 -- Board: SRF3, version: 0
2017-07-21 @ 18:35:46 -- Unique device ID received - 0x3b00244334570720393730

Reading Setup Finished

Main configuration screen does not load

[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1166292/Log.txt)

HighTechPauper, 2017-12-20:
I found that having port peripherals enabled messed up the comm with esc's. Disabled port peripherals (RunCam, SmartAudio, Tramp) and all was well again. So now I just remember to disable the ports mods before I do any esc work

RipperDrone, 2018-01-26:
No, doesn't work here: Reading Setup Finished, but empty screen, no elements of ESC data shown... :-( Have disabled all peripherals like SmartAudio, SmartPort, GPS, ESC telemetry.

closed 2020-11-13 by mikeller