blheli-configurator/blheli-configurator #67: Help!!! Flashed with 16.67RC dshot command.... now nothing works????

Apex808, 2017-07-29:
Plz help!! i flashed my hobbywing xrotor 30A esc with 16.67RC dshot commands and they no longer work no sound from motors and cannot read setup anymore to flash back to 16.6 which was the firm ware they came with. any ideas what happend and how i can get back to 16.6? or are these bricked?

swilkens, 2017-08-04:
Try to flash with the official Blheli Suite software in stead of the blheli-configurator. If you flash them through the flight controller, be sure to set the interface to interface type "E" (SILABS BLHeli Bootloader (Cleanflight)
https://blhelisuite.wordpress.com/

If that doesn't work, you'll have to flash the ESCs using the 4wire interface. This way of flashing requires an Arduino or a FTDI programmer. Here's how to do that, using an Arduino Nano:
https://www.youtube.com/watch?v=86Q7sOXL6t0

HighTechPauper, 2017-12-20:
I found that having port peripherals enabled messed up the comm with esc's. Disabled port peripherals (RunCam, SmartAudio, Tramp) and all was well again. So now I just remember to disable the ports mods before I do any esc work