blheli-configurator/blheli-configurator #69: Autoload firmware for flash don't work

sl1pkn07, 2017-08-05:
[image]

log:

```
Running - OS: <strong>Linux</strong>, Chrome: <strong>62.0.3175.4</strong>, Configurator: <strong>1.0.14</strong>
Last used port wasn't saved "yet", auto-select disabled.
Uncaught Error: Not found chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/libraries/q.js 134 29 Error: Not found
    at Object.callback (chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/flashing_helpers.js:170:29)
    at safeCallbackApply (extensions::uncaught_exception_handler:27:15)
    at handleResponse (extensions::sendRequest:67:7)
Connecting to: /dev/ttyUSB0
SERIAL: Connection opened with ID: 1, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 1
MultiWii API version <span style="color: green">received</span> - <strong>1.36.0</strong>
Flight controller info, identifier: <strong>CLFL</strong>, version: <strong>2.1.0</strong>
Running firmware released on: <strong>Jul 30 2017 13:13:34</strong>
Board: <strong>SRF3</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x2300375233570f20333531</strong>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 144,243,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 14,9,21,9,9,4,255,3,255,9,3,2,1,85,170,1,255,255,255,255,255,4,255,255,255,4,255,40,80,4,255,2,0,122,255,1,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,90,84,87,83,112,80,114,111,50,48,65,80,114,109,35,35,66,76,72,69,76,73,35,70,51,57,48,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 144,243,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 14,9,21,9,9,4,255,3,255,9,3,1,1,85,170,1,255,255,255,255,255,4,255,255,255,4,253,40,80,4,255,2,0,122,255,1,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,90,84,87,83,112,80,114,111,50,48,65,80,114,109,35,35,66,76,72,69,76,73,35,70,51,57,48,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 1
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 144,243,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 14,9,21,9,9,4,255,3,255,9,3,1,1,85,170,1,255,255,255,255,255,4,255,255,255,3,249,40,80,4,255,2,0,122,255,1,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,90,84,87,83,112,80,114,111,50,48,65,80,114,109,35,35,66,76,72,69,76,73,35,70,51,57,48,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 2
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 144,243,99,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 14,9,21,9,9,4,255,3,255,9,3,2,1,85,170,1,255,255,255,255,255,4,255,255,255,4,255,40,80,4,255,2,0,122,255,1,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,90,84,87,83,112,80,114,111,50,48,65,80,114,109,35,35,66,76,72,69,76,73,35,70,51,57,48,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup <span style="color: green">finished</span>
Could not load firmware for ZTW Spider Pro 20A Premium MULTI 14.9: <span style="color: red">File is unavailable</span>
```

DieHertz, 2017-08-05:
This ESC probably has a messed up firmware name on GitHub, you still can flash it providing a HEX explicitly though.

DieHertz, 2017-08-05:
Yeah indeed is.
See [this](https://github.com/bitdump/BLHeli/blob/master/SiLabs/Hex%20files/ZTW_SPIDER_PRO_20A_Premium_MULTI_REV14_9.HEX)? Everything in uppercase but the `Premium` part.
Maybe I'll fix it one day :-)

closed 2020-11-13 by mikeller