blheli-configurator/blheli-configurator #72: Flashing fails ESC showing incorrect name

tmadison-gpsw, 2017-08-30:
`ESC 2: flashing <span style="color: red">failed</span> - <strong>Error: LAYOUT found <span style="color: red">ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ</span>, <strong>#G_H_30#</strong> expected`

[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1265284/Log.txt)

alloy1090, 2017-10-04:
I can confirm the same issue. Adding my log for additional info. I was attempting to flash to 16.63 when this occurred for A_H_25. I've got a 4-in-1 ESC, the one that comes with the wizard x220s. It runs DSHOT and is a 30A. Running windows 7. Note that this has only happened for "one" of the escs on the 4-in-1. The other 3 report the correct name, etc. 
Chrome Version 61.0.3163.100 (Official Build) (64-bit)
BLheli configurator 1.0.14
BF configurator 3.2.1
Was running BF 3.2 RC5 at the time this broke.

[Log2.txt](https://github.com/blheli-configurator/blheli-configurator/files/1354808/Log2.txt)

DieHertz, 2017-10-04:
Tick the "ignore mcu and layout" checkbox upon selecting firmware to flash and try again.
You'd better seek advice at RCGroups if that doesn't help.

alloy1090, 2017-10-04:
@DieHertz Thank you! That did it.

HighTechPauper, 2017-12-20:
I also found that having port peripherals enabled messed up the comm with esc's. Disabled port peripherals (RunCam, SmartAudio, Tramp) and all was well again. So now I just remember to disable the ports mods before I do any esc work

closed 2020-11-13 by mikeller