blheli-configurator/blheli-configurator #75: Configurator not showing any ESCs

jplong74, 2017-09-19:
Betaflight / DYSF4PRO (DYS4) 3.2.0 Sep 11 2017 / 06:33:42 (649ccd7c7) MSP API: 1.36

ESCs = Spedix GS30 (was able to configure via BLHeli Suite)

tailed this from the log file:
SERIAL: Connection opened with ID: 43, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 43
MSP data request timed-out: 1
MSP data request timed-out: 1
Unknown code detected: 110
MSP data request timed-out: 1
MultiWii API version <span style="color: green">received</span> - <strong>1.36.0</strong>
Flight controller info, identifier: <strong>BTFL</strong>, version: <strong>3.2.0</strong>
Running firmware released on: <strong>Sep 11 2017 06:33:42</strong>
Board: <strong>DYS4</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x1f003c3436510b39333434</strong>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>

Jimmi46, 2017-10-20:
[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1402030/Log.txt)

HighTechPauper, 2017-12-20:
I found that having port peripherals enabled messed up the comm with esc's. Disabled port peripherals (RunCam, SmartAudio, Tramp) and all was well again. So now I just remember to disable the ports mods before I do any esc work

closed 2020-11-13 by mikeller