blheli-configurator/blheli-configurator #76: failed to flash 2 out of 4 ESCs and now they are gone

milkypostman, 2017-09-20:
I have *DYS ESC XSD 30A v2 3-5S BLHeli_S* ESCs and I went to flash them with the latest BLHeli_S so that I could use the DSHOT commands. I think it was `16.67`. I did a "Flash All" and one succeeded then one failed and the process stopped. I tried to flash again, just the failed one, but that didn't work either. I tried again on different ones but still it failed. I was able to capture one Log (attached) but now I have two ESCs that are updated and two that are bricked in some way. When I read the config now they don't show up at all in *BLHeli_S Configurator*. 

Is there any way that I would be able to recover these outside of getting access to a windows machine and trying the c2 flash method? I have an Arduino UNO but currently no way to run the *BLHeli Suite*. Looking into options.

[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1316566/Log.txt)

floggi, 2017-09-29:
Have something similar here. I was flashing my "Racerstar 4in1 ESC 20A V2" with newer BLHeli_S. I was doing with BLHeli configurator through FC connection and target was 16.67. After pressing "Flash All" it was able to flash all ESCs. After disconnect (and reboot) 2 of the 4 ESCs started "singing", some kind of beep loop. So final state, 2 ESCs are still working fine but the others are not detected anymore, regardless what I'm doing. Tried to re-flash but can't establish connection.

milkypostman, 2017-10-01:
I created a 3pin connector to do a “hard” flash and that unbricked them.
But not sure if 4-in-1 have these pads.

johnnephin, 2017-10-02:
Keep hitting read and they might show.  I had one that did not flash show back up but two are still missing.

closed 2020-11-13 by mikeller