blheli-configurator/blheli-configurator #78: Connection works but "Read setup" doesn't show anything.

vinz486, 2017-09-23:
Hi,
using this machine http://www.overskyrc.com/warlark-pro-brushless-fpv-whoop-drone-racing-quad-copter-p-776.html
I cannot read ESC configuration: when I click on "read setup" the log shows as "read finished" but nothing happens on configuration window.

Betafligth is 3.2.0. RC5

Log is attached to this post.

Thank you!

[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1326584/BLHELY_conf_with_BF3.2.0RC5.txt)

jplong74, 2017-09-23:
this is what i get as well

DieHertz, 2017-09-23:
Have you powered your ESCs first?

jplong74, 2017-09-23:
I did yes

vinz486, 2017-09-23:
Yes, I tried both with battery and without.

DieHertz, 2017-10-15:
I have the same quad now and don't experience this issue :-(
I may only recommend reflashing ESCs from scratch in BLHeliSuite, usually after that the issue never manifests itself

HighTechPauper, 2017-12-20:
I found that having port peripherals enabled messed up the comm with esc's.  Disabled port peripherals (RunCam, SmartAudio, Tramp) and all was well again.  So now I just remember to disable the ports mods before I do any esc work

mchaos, 2018-08-09:
I am having this issue, and I have disable everything and it still does not work.

TrueExile, 2018-08-20:
I am also having the same issues it seems. 

I have a fresh copy of blheli and am using a emax magnum v2 
Have tried with both peripherals on and off. battery is of coarse plugged in when reading

mchaos, 2018-08-21:
I figured out my issue. I am new to blheli32, and did not realize that the BLHeli configurator does not work with BLheli32. You have to use BLheli32 suite... I am a linux man, so had to do some special stuff to get it to connect in wine, but works good... More options

closed 2020-11-13 by mikeller