blheli-configurator/blheli-configurator #79: Loud clicking noise coming from all of my motors after flashing the ESC

ExogenesisBen, 2017-09-27:
After flashing my ESC for the 20th or so time (while trying to change the start up tune) after one random time the motors started clicking loudly whenever I power the drone. 
I am no longer able to Read the Setup of the drone when I connect it to BLHeli.

Additionally, when I flashed it I set the frequency to particularly low, with this setting:

test1:
	mov	Temp3, #500
	mov Temp4, #1
	mov Temp5, #60
	mov Temp6, #20	
	jmp music

test2:
	mov	Temp3, #500
	mov Temp4, #1
	mov Temp5, #30
	mov Temp6, #20	
	jmp music

test3:
	mov	Temp3, #500
	mov Temp4, #1
	mov Temp5, #30
	mov Temp6, #10	
jmp music
What else can I provide of use to help debug the issue?

ExogenesisBen, 2017-09-27:
[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1338675/Log.txt)

baujahr58, 2017-10-06:
Hello ExogenesisBen,
it is able that I have the same problem with my hoobywing 4in1 esc 40a.
Did you solve the problem?
I`m a total beginner in fpv, 59 years old and fighting with these fucking esc`s.
At the moment I don`t know what to do. 
If you could help me, please send me a comment.
Have a nice day,
Heinz ----- from Duisbrg in germany

ExogenesisBen, 2017-10-15:
I'm afraid I had no success in solving it, sorry!

HighTechPauper, 2017-12-20:
I had the same issue on some esc brands, I was set to DShot600, and when I went back to DShot300/150 Multishot or Oneshot125 the issue went away.  It did not do that on DShot600 before the esc update.  I just run DShot300 on those esc brands and all is well now

closed 2020-11-13 by mikeller