blheli-configurator/blheli-configurator #81: After flashing

baujahr58, 2017-10-03:
Hi guys
My system:
 F4 Flight Controller---http://hobbywing.com/goods.php?id=590&filter_attr=10947.0
4in1 Esc 49a---http://hobbywing.com/goods.php?id=588&filter_attr=10947.0
After switching - Read Setup -nothing happened in the  Blheli-Configurator.
The FC is beeping in staccato.
Look to the screenshot.
[image]
This all happened after 1. time flashing, because at the beginning all was fine!
What can I do?
Please help.
Thank`s,
Heinz

Qwizfpv, 2017-10-10:
same problem

baujahr58, 2017-10-10:
Do you have any idea to solve the problem, Qwizfpv?

baujahr58, 2017-10-10:
I sent a mail to hobbywing.com. When I get the answer, I`ll inform you.

Qwizfpv, 2017-10-12:
no

baujahr58, 2017-10-12:
That‘s bad, but I‘m checking to flash the esc with arduino nano. Without Blheli-Suite. Someone showed it on youtube. And I‘m waiting for a mail from hobbywing.com.
If I flash that esc, I‘ll tell you.
Ok, that‘s it. 
Heinz

HighTechPauper, 2017-11-24:
I seem to have found a way to get my esc's back to 16.66/16.67 dshot.  If you have only 1 esc signal wire hooked to the FC then I was able to see the esc in blheli chrome configurator and reflash it with the older rev.  after doing each one individually and then hooking all signal wires back to FC everything reads normal.  I also tried upgrading to 16.67 release on all 4 before I hooked back up again which was successful and they seem to fly fine but I cannot see them in the configurator anymore,  Looks like I will be wiring them back individually again to downgrade and leave it there.  This is Eachine 20A EMF8BB V1.3 esc's but my E-Max 20A Bullets upgraded t0 16.67 and still read fine.  I had thought it must be these cheap Eachine esc's but now I wonder if its as much a problem with tghe 16.67 rev.  Thoughts?

closed 2020-11-13 by mikeller