blheli-configurator/blheli-configurator #87: Does Not Work with Lumenier ALPHA AIO Flight Controller

mlwellbaum, 2017-11-01:
[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1435126/Log.txt)

Connects to FC but returns no information after reading setup

richard-scott, 2017-11-01:
Are you using the suite from here: http://www.blheli32.com/download/

mlwellbaum, 2017-11-01:
No. I’m using the Chrome Blheli Configurator.

Mike

richard-scott, 2017-11-02:
The Chrome Blheli Configurator doesn't work with Blheli32 ESC's.

DzikuVx, 2017-11-02:
@richard-scott it will probably never work with BLHeli32 and @DieHertz stated that multiple times.

@mlwellbaum this is rather not Configurator issue, but software that is running this flight controller

mlwellbaum, 2017-11-02:
Good to know...thanks!

Mike

yattiyada, 2018-02-05:
so will it EVER support Blheli_32?

closed 2020-11-13 by mikeller