blheli-configurator/blheli-configurator #90: Unable to connect -

BigNickman, 2017-11-14:
Hi,

I'm using the DYS F4 FC and 4in1 DYS DSHOT 600 30A ESC.  I'm not able to connect to BLHELI Suite or the Chrome Configurator.  I've gone through the driver troubleshooting and have no issues finding a the correct com port.  I'm able to connect without issues in BF, and I can spin up the motors without issues in BF.  I just can seem to get into BLHELI suite.  See attached Log from my attempt to connect to Chrome Configurator.  When I try to connect through Windows version of BLHELI suite I get this error, "connection to fc failed check port and power".  I have nothing wired to UART 1 on the FC, which I read is the USB UART.  Any help/knowledge would be great.

Thanks!

[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1471711/Log.txt)

YanikTheYak, 2017-12-14:
Hi
I'm new to this code. But it looks like your USB is connecting to the FC, but the pass through to the ESCs is not happy. The Init that is failing is the communication to the ESCs. 
This will probably make no difference since it is the FC that passes the messages to the ESC (probably at its own speed), but have you tried a lower baud rate?

Ian

Indy6, 2018-02-27:
Hi, 
No news here?
I've experiencing exactly the same issue. I can fly, I can connect BF configurator, but I can't connect BLHELI configurator on mac and PC or BLHELI suite on PC.
I tried to update the FC firmware as well as downgrade... nothing seems to work...
Any idea?
Thanks
[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1763168/Log.txt)

BigNickman, 2018-02-27:
Think it was a hardware issue with my fc.  I gave up on it a while back.

evans036, 2018-05-19:
i had the same problem (using dys f4 and another controller).

but this does work: http://www.blheli32.com/download/

dunno why

steve

closed 2020-11-13 by mikeller