blheli-configurator/blheli-configurator #91: unable to connect  log file as below

metalbits1966, 2017-11-18:


metalbits1966, 2017-11-18:
Running - OS: <strong>MacOS</strong>, Chrome: <strong>62.0.3202.94</strong>, Configurator: <strong>1.0.14</strong>
Running - OS: <strong>MacOS</strong>, Chrome: <strong>62.0.3202.94</strong>, Configurator: <strong>1.0.14</strong>
Last used port wasn't saved "yet", auto-select disabled.
Uncaught Error: Not found chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/libraries/q.js 134 29 Error: Not found
    at Object.callback (chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/flashing_helpers.js:170:29)
    at safeCallbackApply (extensions::uncaught_exception_handler:27:15)
    at handleResponse (extensions::sendRequest:67:7)
PortHandler - Found: /dev/tty.usbmodemFD1421,/dev/cu.usbmodemFD1421
Connecting to: /dev/tty.usbmodemFD1421
SERIAL: Connection opened with ID: 25, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 25
MultiWii API version <span style="color: green">received</span> - <strong>1.31.0</strong>
Flight controller info, identifier: <strong>BTFL</strong>, version: <strong>3.1.7</strong>
Running firmware released on: <strong>May  2 2017 19:59:41</strong>
Board: <strong>MKF4</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x2c00223436510b33323835</strong>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>
[... 39 lines cut ...]
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,102,0
ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>

DieHertz, 2017-11-18:
What ESCs? Please note this GitHub repository is not a place for user support, seek support at RCGroups or a similar forum.

metalbits1966, 2017-11-18:
hi thanks the ecs,s are little bee summer 35a with blheli32

DieHertz, 2017-11-18:
There's no advertised blheli32 support.

closed 2017-11-18 by DieHertz