blheli-configurator/blheli-configurator #92: Helix with 20am little bee escs

cdsalty, 2017-11-21:
I am trying to simply flash my escs and change motor direction. Can you please offer me some assistance?

2017-11-21 @ 00:24:55 -- Running - OS: Windows, Chrome: 62.0.3202.94, Configurator: 1.0.14
2017-11-21 @ 00:25:15 -- Serial port successfully opened with ID: 1
2017-11-21 @ 00:25:16 -- MultiWii API version received - 1.36.0
2017-11-21 @ 00:25:17 -- Flight controller info, identifier: BTFL, version: 3.2.2
2017-11-21 @ 00:25:18 -- Running firmware released on: Nov 11 2017 00:31:36
2017-11-21 @ 00:25:20 -- Board: IMF3, version: 0
2017-11-21 @ 00:25:21 -- Unique device ID received - 0x24001c5433571320333536
2017-11-21 @ 00:25:25 -- Connection failed, ensure that you have selected the correct port or restart your device.

richard-scott, 2017-11-21:
What are you using to connect to them with? Your Flight Controller? If so, which one?
Also, do you have a product link for the ESC's.

cdsalty, 2017-11-22:
You guys are amazing as I never thought I would hear from anyone. I watched
more and more tutorials and finally figured it out. Much Thanks!

Christopher Soltis

Nickstash, 2018-02-26:
Hey Christopher.  I know this was a few months ago, but I've run in to the exact same issue.  What did you do to resolve it?

cdsalty, 2018-02-26:
Be glad to help but what forum or what was the original question?  Hope to
hear back...

Christopher Soltis

Nickstash, 2018-02-26:
It was your post about not being able to use blheli configurator with your helix. Connection was failing.

Nickstash, 2018-02-26:
I am trying to simply flash my escs and change motor direction. Can you please offer me some assistance?

That was your original question.

cdsalty, 2018-02-26:
Here is the quick rundown...

Go to motor tab, take off props and click you took props off.  Move motor
slider all the way to the top and then... plug in your battery and wait for
the beeps to stop. Them slide the motor tab all the way down and wait for
your final beeps. Your esc's are set.

Motor direction, simply swap any two wires soldered to your motor.  Sounds
silly but that will fix it and you're good to go.

Christopher Soltis

Nickstash, 2018-02-26:
Ok so you didn't actually get blheli passthrough to work on your helix
then...  You shouldn't need to do that old school calibration with dshot
enabled.

cdsalty, 2018-02-26:
I just honestly don't recall... it won't run dshot so I fly other qwads.

Christopher Soltis

DieHertz, 2018-02-26:
Afaik this target has output buffers, hence bidirectional communication on it is impossible just like the KISS FC

Nickstash, 2018-02-26:
I ended up using the blheli suite and it worked fine.

DieHertz, 2018-02-26:
Then BLHeli Configurator should work just as well now

Nickstash, 2018-02-26:
It doesnt

DieHertz, 2018-02-26:
Please provide a log from both BLHeliSuite and BLHeli Configurator then

closed 2020-11-13 by mikeller