blheli-configurator/blheli-configurator #94: Would it be possible to have a sync button in the Configurator

Fpv-Strafer, 2017-12-02:
Right now when I do the first read through Blheli of the Esc's they possibly come up in the first box saying they are not synced and the only way I know of to Sync them is to change something insignificant (beeper timer) and do a write to all 4 esc's.
Would it be possible to have a button put in the configurator that would sync up the esc.s automatically?
Or is there another way that the esc's can be synced that I don't know of.

Fpv-Strafer, 2017-12-02:
Another Question if some one could explain or answer for me.
With Dshot everyone says that you don't have to calibrate or make sure the esc's are synced.
Ive seen on a few occasions where this didn't hold true and doing a cal cured the issues.
So why does Dshot not require a calibration or to be synced (By doing a write to all 4 esc's) ?

richard-scott, 2017-12-02:
It's not that the 4 ESC's all need to be in sync with each other, but the sync is with the signal from the flight controller.

Normally the lack of a signal to sync with is due to a low throttle signal at minimum throttle stick movement so increasing that reduces the desync issues.

Dshort can suffer the same problem if the min idle % is too low. So in short, no you don't need a "sync button" you just need to configure your low throttle signal high enough to be above the minimum signal the ESC's expect.

DieHertz, 2017-12-02:
"Not in sync" usually pertains to parameters like "motor timing" or "beep strength", not per-ESC calibration values.
While it's possible to have a "sync" button, what would be the common denominator? Imagine your ESC 1&2 are at Medium timing, while others are at MediumHigh, which timing should be set to all ESCs during sync?

DieHertz, 2017-12-02:
Calibration has no effect on DSHOT, especially in flight. It could only have effect during FC initialization phase before correct protocol auto-detection, once BLHeli_S has detected DSHOT as it's protocol, calibration values no longer matter and have 0 effect.

closed 2018-04-22 by Fpv-Strafer