blheli-configurator/blheli-configurator #104: 6a esc ph10 #1 of 4 error 255.255

skottlotz, 2017-12-26:
chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/libraries/polyfill.min.js:1:27885

[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1587823/Log.txt)

closed 2017-12-26 by DieHertz

skottlotz, 2017-12-26:
closed? i need help!

skottlotz, 2017-12-26:
So. What's the answer?

  error 255.255 (#104) 
Closed #104.

DieHertz, 2017-12-28:
This repo for development, not a place to drop your one-liner issue and get immediate help.
Tick the "Ignore inappropriate MCU and Layout" checkbox while flashing, just as the error message suggests you, if it doesn't help - forward your question to RCGroups.
[image]