blheli-configurator/blheli-configurator #106: Im stuck at the warning screen

ashkan1244, 2017-12-28:
After I press connect the only thing I see is this: https://imgur.com/a/JEfaA
I have tried pressing "read setup", but nothing happens

richard-scott, 2017-12-28:
Did you connect your flight battery before clicking read?

sandav, 2018-01-04:
I got the same problem. blank screen after read setup.
Battery is plugged in, reading seems to finish allright but no data is shown in the configurator screen.
All is working fine, I can start the motors with my remote but I need to reverse direction on two motors.
I hope it is just a small mistake I make but any help would be appreciated.

richard-scott, 2018-01-04:
What ESC's are you using? Do you have a link for what you purchased?

sandav, 2018-01-04:
The last working time was with these esc's. I was able to read, upgrade and modify the firmware...
https://www.banggood.com/4X-Racerstar-RS20A-20A-BLHELI_S-OPTO-2-4S-ESC-Support-Oneshot42-Multishot-for-FPV-Racing-p-1064324.html?rmmds=myorder

I replaced them with these to benefit from dshot1200, With these esc's, I have the exact same issue as ashkan1244. Read seems to go allright but after that, I am presented wit a blank screen.
https://www.banggood.com/4-PCS-Racerstar-Tattoo-30A-BLheli_32-2-4S-Brushless-ESC-Dshot1200-for-Racing-Drone-p-1219847.html?rmmds=myorder&cur_warehouse=CN

DieHertz, 2018-01-04:
Which version of app do you have?

sandav, 2018-01-04:
app version 1.1.0

DieHertz, 2018-01-04:
Then apart from the blank screen your should also get a message saying "Blheli32 is not supported"

sandav, 2018-01-04:
nope, just tried again and output is the same.
[image]

sandav, 2018-01-04:
[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/1604058/Log.txt)

Attached is the blheli debug log. A lot of errors and warnings but it is beyond my knowledge to see what I could fix.

DieHertz, 2018-01-04:
No BLHeli32 support

DieHertz, 2018-01-04:
This app is not discontinued, and won't be anytime soon.

sandav, 2018-01-04:
Thanks for the time and swift replies.
When Google apps will stay is good news cause I like the app a lot (as long as it is working) but Google states otherwise.
[Google will end support for Chrome apps on Windows, Mac, and Linux](https://www.theverge.com/2016/8/19/12555052/google-shutting-down-chrome-apps)
I guess this is explained many times but I missed it but if you're telling me that you are here to stay, I'm happy.

DieHertz, 2018-01-04:
It's already explained on the release page, just have a look:
https://github.com/blheli-configurator/blheli-configurator/releases/tag/1.1.0

mdmbc, 2018-01-04:
With how often BLHeli32 related issues are submitted, would it make sense to include a more clear description/warning on the main page of the app about the lack of support for BLHeli32?

DieHertz, 2018-01-04:
There's a message upon trying to operate BLHeli 32 ESCs, but it's not working for some reason. I'll fix it tomorrow.
Also it's mentioned in the readme I think.

ajspera, 2018-09-16:
Getting this same problem on OSX standalone with Blheli_s 35A 32bit ESC and Omnibus F3 FC

closed 2020-11-13 by mikeller