blheli-configurator/blheli-configurator #108: I can't connect to ESC via standalone application while chrome version can.

Sergey-Afonin, 2018-01-04:
2018-01-04 @ 20:16:42 -- Running - OS: Windows, Chrome: 63.0.3239.108, Configurator: 1.1.0
2018-01-04 @ 20:16:49 -- Serial port successfully opened with ID: 1
2018-01-04 @ 20:16:49 -- MultiWii API version received - 1.31.0
2018-01-04 @ 20:16:49 -- Flight controller info, identifier: BTFL, version: 3.1.6
2018-01-04 @ 20:16:49 -- Running firmware released on: Feb 21 2017 14:35:09
2018-01-04 @ 20:16:49 -- Board: PIKO, version: 0
2018-01-04 @ 20:16:49 -- Unique device ID received - 0x3a00295533570620313836
2018-01-04 @ 20:16:59 -- Connection failed, ensure that you have selected the correct port or restart your device.
2018-01-04 @ 20:17:11 -- Serial port successfully opened with ID: 2
2018-01-04 @ 20:17:13 -- MultiWii API version received - 1.31.0
2018-01-04 @ 20:17:15 -- Flight controller info, identifier: BTFL, version: 3.1.6
2018-01-04 @ 20:17:15 -- Running firmware released on: Feb 21 2017 14:35:09
2018-01-04 @ 20:17:15 -- Board: PIKO, version: 0
2018-01-04 @ 20:17:15 -- Unique device ID received - 0x3a00295533570620313836
2018-01-04 @ 20:17:21 -- Connection failed, ensure that you have selected the correct port or restart your device.
2018-01-04 @ 20:17:33 -- Serial port successfully opened with ID: 3
2018-01-04 @ 20:17:43 -- Connection failed, ensure that you have selected the correct port or restart your device.
2018-01-04 @ 20:18:10 -- Serial port successfully opened with ID: 4
2018-01-04 @ 20:18:10 -- MultiWii API version received - 1.31.0
2018-01-04 @ 20:18:12 -- Flight controller info, identifier: BTFL, version: 3.1.6
2018-01-04 @ 20:18:12 -- Running firmware released on: Feb 21 2017 14:35:09
2018-01-04 @ 20:18:13 -- Board: PIKO, version: 0
2018-01-04 @ 20:18:13 -- Unique device ID received - 0x3a00295533570620313836
2018-01-04 @ 20:18:20 -- Connection failed, ensure that you have selected the correct port or restart your device.
-----------------------------------------------------------------------------------------------------
Chrome version says and works fine:

2018-01-04 @ 20:24:15 -- Running - OS: Windows, Chrome: 63.0.3239.108, Configurator: 1.1.0
2018-01-04 @ 20:24:18 -- Serial port successfully opened with ID: 4
2018-01-04 @ 20:24:18 -- MultiWii API version received - 1.31.0
2018-01-04 @ 20:24:20 -- Flight controller info, identifier: BTFL, version: 3.1.6
2018-01-04 @ 20:24:21 -- Running firmware released on: Feb 21 2017 14:35:09
2018-01-04 @ 20:24:21 -- Board: PIKO, version: 0
2018-01-04 @ 20:24:21 -- Unique device ID received - 0x3a00295533570620313836

rdlkgliders, 2018-01-13:
I get this too! sometimes I connect but usually not. Happens on multiple F.C's Piko, Omnbus 4 in 1, etc..

pierre-quelin, 2018-02-20:
It seems the same. BetaFlight standalone connection ok on linux.
Blheli-configurator connection ok via chrome on Windows 7.

- BLHeli-Configurator_linux64_1.1.0
- Debian 9 arch=amd64

- Log
[8187:8212:0220/192740.162191:ERROR:nss_util.cc(724)] After loading Root Certs, loaded==false: NSS error code: -8018
ATTENTION: default value of option force_s3tc_enable overridden by environment.
[8187:8212:0220/192743.579677:ERROR:serial_io_handler.cc(149)] Failed to open serial port: FILE_ERROR_ACCESS_DENIED

richard-scott, 2018-02-20:
ok, asking the obvious first... are you closing/disconnecting other applications before using the configurator... only one app can open the serial port at any one time.

pierre-quelin, 2018-02-20:
Yes of course. I'm in the dialout group. It's working fine for beta-flight
configurator

mrtwizta, 2018-02-28:
2018-02-28 @ 20:27:36 -- Running - OS: Windows, Chrome: 63.0.3239.132, Configurator: 10.2.0

2018-02-28 @ 20:27:39 -- Loaded release information for configurator from GitHub.

2018-02-28 @ 20:39:44 -- Serial port successfully opened with ID: 1

2018-02-28 @ 20:39:54 -- No configuration received within 10 seconds, communication failed

2018-02-28 @ 20:39:54 -- Serial port successfully closed

pierre-quelin, 2018-02-28:
Solved for me. Sorry, all I had to do was to be patient (10s).

woutput, 2018-03-31:
I have the same issue. Also note that after the last message shown by @mrtwizta and @Sergey-Afonin retrying to connect fails earlier because it can't connect anymore. Only removing power from FC & ESCs and reconnecting brings it back to the first failing message.
@pierre-quelin please explain the exact steps how to work around this issue. I don't know how.

regnets, 2018-05-02:
Same for me. If you need any further information, i would be happy to provide them.

FCs: 
Omnibus F3
Omnibus F3 Pro
ESCs: 
Anniversary Special Edition Racerstar REV35 35A
Racerstar RS20Ax4 20A 4 in 1

closed 2020-11-13 by mikeller