blheli-configurator/blheli-configurator #124: Attempting to enter BLHeli Config after flashing to latest 16.7 official version

rickcote0041, 2018-02-16:
2018-02-16 @ 18:51:25 -- Running - OS: Windows, Chrome: 64.0.3282.167, Configurator: 1.1.0

2018-02-16 @ 18:51:27 -- Serial port successfully opened with ID: 1

2018-02-16 @ 18:51:28 -- MultiWii API version received - 1.36.0

2018-02-16 @ 18:51:29 -- Flight controller info, identifier: BTFL, version: 3.2.5

2018-02-16 @ 18:51:30 -- Running firmware released on: Feb 11 2018 00:46:59

2018-02-16 @ 18:51:31 -- Board: OMNI, version: 0

2018-02-16 @ 18:51:32 -- Unique device ID received - 0x37001e4734570d20383631

2018-02-16 @ 18:51:37 -- Connection failed, ensure that you have selected the correct port or restart your device.

Sqwikken, 2018-02-27:
I have this same failure with the same target, though I'm only trying to connect to my ESCs for the first time since updating to 3.2.5:

2018-02-26 @ 22:23:02 -- Running - OS: Windows, Chrome: 63.0.3239.108, Configurator: 1.1.0

2018-02-26 @ 22:23:12 -- Serial port successfully opened with ID: 1

2018-02-26 @ 22:23:13 -- MultiWii API version received - 1.36.0

2018-02-26 @ 22:23:15 -- Flight controller info, identifier: BTFL, version: 3.2.5

2018-02-26 @ 22:23:16 -- Running firmware released on: Feb 11 2018 00:46:59

2018-02-26 @ 22:23:17 -- Board: OMNI, version: 0

2018-02-26 @ 22:23:18 -- Unique device ID received - 0x3200274234571120363434

2018-02-26 @ 22:23:22 -- Connection failed, ensure that you have selected the correct port or restart your device.

rickcote0041, 2018-02-27:
Silly mistake in my scenario.I discovered that using a different fully
charged 2s battery fixed my issue... The 1s battery I was attempting to use
earlier was not enough voltage to power the 4 in one esc. I thought the esc
worked with 1s or 2s ...after a careful review of the esc doc I found out
it only supports 2s... I hope this update is helpful to someone. Happy
flying all!!

woutput, 2018-03-31:
Same observation as #108, #131 and #139 so maybe the same root cause; additional diagnostic information in the log might help debugging.

regnets, 2018-05-02:
Same for me. If you need any further information, i would be happy to provide them.

FCs: 
Omnibus F3
Omnibus F3 Pro
ESCs: 
Anniversary Special Edition Racerstar REV35 35A
Racerstar RS20Ax4 20A 4 in 1

regnets, 2018-05-05:
I just wanted to give you an update.

The Connection is completely without issue with Betaflight 3.0.1.

So as a workarround, just downgrade your FC to 3.0.1 do your BL-Heli Configuration and reflash to 3.3.1.

I will open an issue in the Betaflight Repo.

For convenience, can we just merge these three issues?

closed 2020-11-13 by mikeller