blheli-configurator/blheli-configurator #127: click read set up and nothing happends????

Cleevm, 2018-02-22:
when i plug into blheli it connects just fine but when i click read set up it dosent take me to the main configuration screen help?

DieHertz, 2018-02-22:
Which ESCs?

Cleevm, 2018-02-22:
blheli dys 35a

DieHertz, 2018-02-22:
Could you show a product link?

Cleevm, 2018-02-22:
[Log help.txt](https://github.com/blheli-configurator/blheli-configurator/files/1749186/Log.help.txt)

Cleevm, 2018-02-22:
https://www.banggood.com/DYS-BLHeli_32bit-35A-35amp-Brushless-ESC-2-6S-Dshot1200-Ready-For-FPV-Racing-p-1148598.html?gmcCountry=US&currency=USD&createTmp=1&utm_source=googleshopping&utm_medium=cpc_ods&utm_content=heath&utm_campaign=pla-multi-us-pc&gclid=CjwKCAiA8bnUBRA-EiwAc0hZk5vZFu4ZwPiLUfNaZdN0xTRJxrZD9wjUPNOCj30Ooc39FmLj9pYOURoCBG8QAvD_BwE&cur_warehouse=HK

DieHertz, 2018-02-22:
Your ESC is BLHeli32, BLHeli32 is a closed-source firmware and unfortunately it's not supported

closed 2018-02-22 by DieHertz

Cleevm, 2018-02-22:
ok thanks

Cleevm, 2018-02-22:
is there a configurator that will work with the ecs?