blheli-configurator/blheli-configurator #131: Unable to connect to ESCs

alphat5, 2018-03-01:
Hi guys,

I had this issues a long time ago. When I try to connect to my escs it gives me this error (see logs lower). And some time I could connect to my escs and some time not. I've looked in the issues reports and it's a known issues sadly 😢.

- FC Betaflight F3 (3.2.1)
- ESC T-motor F30A (blheli v16.7)

Here is my log

```
2018-03-01 @ 17:24:01 -- Serial port successfully opened with ID: 23
2018-03-01 @ 17:24:03 -- MultiWii API version received - 1.36.0
2018-03-01 @ 17:24:05 -- Flight controller info, identifier: BTFL, version: 3.2.1
2018-03-01 @ 17:24:05 -- Running firmware released on: Oct 15 2017 19:28:25
2018-03-01 @ 17:24:06 -- Board: BFF3, version: 0
2018-03-01 @ 17:24:07 -- Unique device ID received - 0x390034463457052039****

```
_Unique devinci ID is hiden volontary_ 😉

Thanks in advance for your help :)

woutput, 2018-03-31:
This appears to be equal to (or at least same observation as) #108 and #131

regnets, 2018-05-02:
Same for me. If you need any further information, i would be happy to provide them.

FCs: 
Omnibus F3
Omnibus F3 Pro
ESCs: 
Anniversary Special Edition Racerstar REV35 35A
Racerstar RS20Ax4 20A 4 in 1

regnets, 2018-05-05:
I just wanted to give you an update. 

The Connection is completely without issue with Betaflight 3.0.1.

So as a workarround, just downgrade your FC to 3.0.1 do your BL-Heli Configuration and reflash to 3.3.1.

I will open an issue in the Betaflight Repo.

For convenience, can we just merge these three issues?

closed 2020-11-13 by mikeller