blheli-configurator/blheli-configurator #139: Connection failed

RaffCA, 2018-03-20:
Hello,

Please help with the following problem, trying to connect my F3 3in1 to blheli. Battery is plugged BF is disconnected and I have a message that _'Connection failed, ensure that you have selected the correct port or restart your device'_. Restarted with no luck. Port is the same what I'm using when connecting to BF with no problem.

Thank you for your help.

Please find log below:

_2018-03-20 @ 11:13:34 -- Running - OS: Windows, Chrome: 63.0.3239.108, Configurator: 1.1.0
2018-03-20 @ 11:15:12 -- Serial port successfully opened with ID: 1
2018-03-20 @ 11:15:13 -- MultiWii API version received - 1.37.0
2018-03-20 @ 11:15:15 -- Flight controller info, identifier: BTFL, version: 3.3.0
2018-03-20 @ 11:15:15 -- Running firmware released on: Mar 2 2018 03:56:08
2018-03-20 @ 11:15:16 -- Board: OMNI, version: 0
2018-03-20 @ 11:15:17 -- Unique device ID received - 0x3b001d4d34570e20353734
2018-03-20 @ 11:15:22 -- Connection failed, ensure that you have selected the correct port or restart your device_

richard-scott, 2018-03-20:
What ESCs are you using?

RaffCA, 2018-03-20:
Hi Richard, 
It is an orig Leader 120 with 28A BLHELI_S  2-4S 4in1 ESC Dshot600.
Any suggestions?

richard-scott, 2018-03-20:
Can you configure BF Ok as it will use the same port?

RaffCA, 2018-03-20:
Yes, I can use BF without problem with the same port.

richard-scott, 2018-03-20:
Try and re-flash BetaFlight and see if you can communicate with the ESCs before configuring anything else. Don't forget to run "defaults & save" in the CLI after flashing.

woutput, 2018-03-31:
This appears to be equal to (or at least same observation as) https://github.com/blheli-configurator/blheli-configurator/issues/108

regnets, 2018-05-02:
Same for me. If you need any further information, i would be happy to provide them.

FCs: 
Omnibus F3
Omnibus F3 Pro
ESCs: 
Anniversary Special Edition Racerstar REV35 35A
Racerstar RS20Ax4 20A 4 in 1

regnets, 2018-05-05:
I just wanted to give you an update. 

The Connection is completely without issue with Betaflight 3.0.1.

So as a workarround, just downgrade your FC to 3.0.1 do your BL-Heli Configuration and reflash to 3.3.1.

I will open an issue in the Betaflight Repo.

RaffCA, 2018-05-05:
Hi Andreas, thank you for the input. For the moment I worked it around via re-soldering my motors, since all I wanted in config is to change direction:)

woutput, 2018-05-05:
I doubt if we should merge the three. It appears as if different root causes lead to this same observation.
I would prefer to change the configurator to be more descriptive in error messages. These would then hopefully give more insight when searching for a root cause.

Ask1me, 2018-07-03:
same problem here with FC crazybeeF3 Frysky version, Firmware : betaflight_3.3.3_CrazybeeF3FR (stable)

this what i already try :
1.already tested to update BF to 3.4.0_RC5 and also try to downgrade BF to 3.3.1
2.I try both app : using latest BLHeli configurator standalone and lates BLHeli Suite standalone
3. now it can't bind with my multiprotocol Tx jumper T8SG V2+ :(

is there any suggestions my friend?

this my log file form latest BLHeli Suite 
2018-07-03 @ 22:32:00 -- Running - OS: Windows, Chrome: 63.0.3239.108, Configurator: 1.1.0

2018-07-03 @ 22:32:13 -- Serial port successfully opened with ID: 1

2018-07-03 @ 22:32:13 -- MultiWii API version received - 1.38.0

2018-07-03 @ 22:32:15 -- Flight controller info, identifier: BTFL, version: 3.3.3

2018-07-03 @ 22:32:16 -- Running firmware released on: May 31 2018 12:58:28

2018-07-03 @ 22:32:17 -- Board: CBFR, version: 0

2018-07-03 @ 22:32:18 -- Unique device ID received - 0x2b002a4734571020323638

2018-07-03 @ 22:32:23 -- Connection failed, ensure that you have selected the correct port or restart your device.

closed 2020-11-13 by mikeller