blheli-configurator/blheli-configurator #151: node_contextify.cc:629: Assertion `args[1]->IsString()' failed

richard-scott, 2018-05-26:
When building on a freshly installed system with Node 10 I get this error:

```
C:\users\richard\git\blheli-configurator> gulp apps --win64
gulp apps --win64[2812]: src\node_contextify.cc:629: Assertion `args[1]->IsString()' failed.
 1: node::DecodeWrite
 2: node::DecodeWrite
 3: uv_loop_fork
 4: v8::internal::interpreter::BytecodeDecoder::Decode
 5: v8::internal::RegExpImpl::Exec
 6: v8::internal::RegExpImpl::Exec
 7: v8::internal::RegExpImpl::Exec
 8: 0000020E84384281
```

Apparently its a [bug ](https://github.com/nodejs/node/issues/20325) in node, and can be worked around by installing 'natives'

```
npm install natives
```

mribichich, 2018-06-05:
Take a look at this comment:

nguyenthemanh2601, 2018-06-25:
I encountered the same error when I run the gulp command:

> mannguye@DVN01000156 MINGW64 /e/project/myskills1/myskills/backend/htdocs (master)
> $ gulp
> [13:58:46] Working directory changed to E:\project\myskills1\myskills
> gulp[6476]: src\node_contextify.cc:629: Assertion `args[1]->IsString()' failed.
>  1: node::DecodeWrite
>  2: node::DecodeWrite
>  3: uv_loop_fork
>  4: v8::internal::interpreter::BytecodeDecoder::Decode
>  5: v8::internal::RegExpImpl::Exec
>  6: v8::internal::RegExpImpl::Exec
>  7: v8::internal::RegExpImpl::Exec
>  8: 000003FA44F04281

i try with [https://github.com/nodejs/node/issues/20285#issuecomment-387612673](url)
but, it is not working!
Help me please!

> $ node -v
v10.3.0
> $ npm -v
6.1.0
$ gulp -v
[14:08:55] CLI version 2.0.1
[14:08:55] Local version 4.0.0-alpha.3

>

kanthmikkili, 2018-08-22:
i have tried npm install natives and it worked for me

danybr0, 2018-09-29:
I get this error:

internal/modules/cjs/loader.js:583
    throw err;
    ^

Error: Cannot find module 'gulp-sass'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
    at Function.Module._load (internal/modules/cjs/loader.js:507:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (C:\Users\Dan.Covic\Desktop\github repos\bs4-template-init\gulpfile.js:3:19)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
PS C:\Users\Dan.Covic\Desktop\github repos\bs4-template-init> gulp
internal/modules/cjs/loader.js:583
    throw err;
    ^

Error: Cannot find module 'gulp-sass'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
    at Function.Module._load (internal/modules/cjs/loader.js:507:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (C:\Users\Dan.Covic\Desktop\github repos\bs4-template-init\gulpfile.js:3:19)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)

danybr0, 2018-09-29:
Solved:

´rm -r node_modules
npm install´

`rm -r node_modules
npm install`

HaythemJ, 2018-09-30:
This is worked for me also:

`npm install natives`

vinipuhpuh, 2018-10-31:
`npm install natives`

worked here.

closed 2020-11-13 by mikeller