blheli-configurator/blheli-configurator #155: betaflight_3.3.3_CrazybeeF3FR can't connect to BLHELI configurator, please help me

Ask1me, 2018-07-03:
2018-07-03 @ 22:32:00 -- Running - OS: Windows, Chrome: 63.0.3239.108, Configurator: 1.1.0

2018-07-03 @ 22:32:13 -- Serial port successfully opened with ID: 1

2018-07-03 @ 22:32:13 -- MultiWii API version received - 1.38.0

2018-07-03 @ 22:32:15 -- Flight controller info, identifier: BTFL, version: 3.3.3

2018-07-03 @ 22:32:16 -- Running firmware released on: May 31 2018 12:58:28

2018-07-03 @ 22:32:17 -- Board: CBFR, version: 0

2018-07-03 @ 22:32:18 -- Unique device ID received - 0x2b002a4734571020323638

2018-07-03 @ 22:32:23 -- Connection failed, ensure that you have selected the correct port or restart your device.

2018-07-03 @ 22:32:32 -- Serial port successfully opened with ID: 2

2018-07-03 @ 22:32:42 -- Connection failed, ensure that you have selected the correct port or restart your device.

2018-07-03 @ 22:32:53 -- Serial port successfully opened with ID: 3

2018-07-03 @ 22:32:55 -- MultiWii API version received - 1.38.0

2018-07-03 @ 22:32:57 -- Flight controller info, identifier: BTFL, version: 3.3.3

2018-07-03 @ 22:32:57 -- Running firmware released on: May 31 2018 12:58:28

2018-07-03 @ 22:32:58 -- Board: CBFR, version: 0

2018-07-03 @ 22:32:59 -- Unique device ID received - 0x2b002a4734571020323638

2018-07-03 @ 22:33:03 -- Connection failed, ensure that you have selected the correct port or restart your device.

2018-07-03 @ 22:33:07 -- Serial port successfully opened with ID: 4

2018-07-03 @ 22:33:17 -- Connection failed, ensure that you have selected the correct port or restart your device.

2018-07-03 @ 22:33:31 -- Serial port successfully opened with ID: 5

2018-07-03 @ 22:33:33 -- MultiWii API version received - 1.38.0

2018-07-03 @ 22:33:35 -- Flight controller info, identifier: BTFL, version: 3.3.3

2018-07-03 @ 22:33:36 -- Running firmware released on: May 31 2018 12:58:28

2018-07-03 @ 22:33:37 -- Board: CBFR, version: 0

2018-07-03 @ 22:33:37 -- Unique device ID received - 0x2b002a4734571020323638

2018-07-03 @ 22:33:41 -- Connection failed, ensure that you have selected the correct port or restart your device.

2018-07-03 @ 22:33:49 -- Serial port successfully opened with ID: 6

2018-07-03 @ 22:33:59 -- Connection failed, ensure that you have selected the correct port or restart your device.

already try to reconnect it many times

but its fine with BF configurator

what can i do?

Ask1me, 2018-07-10:
so you mean the steps are like this: 
1. connect usb to PC 
2. open BLHeli suite / configurator 
3. klik connect, and immediately connect lipo? 

Is it already correct?

is there any provision should use which version of BLHelisuite/ configurator?

sorry for my newbie questions.. i really confused, till know still not connected

Joffa86, 2018-07-11:
1.Connect usb to quad
2.connect to blheli
3.plug in your lipo 
4.press read setup.
 I used latest version standalone, not through chrome app, but either should work.

closed 2020-11-13 by mikeller