blheli-configurator/blheli-configurator #160: BLHELI won't read the esc's ?

Flytime1958, 2018-08-24:
I'm having a problem that BLheli won't read set up? Incoming port is ok when connecting . Powering it up with the battery ans then the tab read set-up it only says finished and there is nothing on the screen.

bigey, 2018-09-16:
Connection is ok but failed to read ESC.
```
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 6,51,104,4
sending cmd_DeviceReadEEprom 0 112
received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0
received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0
ESC 1 read settings failed cmd_DeviceReadEEprom ACK_I_INVALID_CMD
```

[complete debug log here](https://github.com/blheli-configurator/blheli-configurator/files/2386460/Log.txt)

Any ideas ?

DieHertz, 2018-09-16:
You have blheli 32, and it's not supported. You'll have greater luck using Wine

closed 2018-09-16 by Flytime1958

bigey, 2018-09-17:
Thank you @DieHertz. I was able to read ESC using BLHELI suite running on Linux with Wine!