blheli-configurator/blheli-configurator #187: Motor Poles min value behiavour

Asizon, 2020-11-27:
ESC: Ifllight succesx 50A
ESC version: AM32-1.58
Configurator: 1.3.0 Nightly
Motor poles have a minimum value of 0, but only at the beginning. When you want to configure this slider, the minimum value stops being 0 and becomes 2. Here an example screenshot:

[image]

Asizon, 2020-12-11:
Seems to be related with the AM32 defaults, this 1.58 firmware comes with motor poles 0 value and it allows, but in BlheliConfigurator we have default min value of 2. @mikeller do you think that can be a problem?

mikeller, 2020-12-12:
I was not able to reproduce this. I suspect that this is misbehaving due to the version of the boot loader on your ESCs not yet supporting the reset command.

You can try to update the boot loader by flashing this 'firmware' and then disconnecting in BLHeli Configurator while the ESC is powered, and waiting for 20 seconds or so: 
[BOOTLOADER_UPDATER_PB4.hex.zip](https://github.com/blheli-configurator/blheli-configurator/files/5682126/BOOTLOADER_UPDATER_PB4.hex.zip)
(**iFlight ESC only**)
After applying this, you will have to flash actual firmware again.

Asizon, 2020-12-12:
This motor poles value comes from the old bootloader and firm yes, I have solved it ussing Defaluts Reset in Blheli Configurator, after this the min value is always 2.

Asizon, 2020-12-22:
Closing this issue because is just an old bootloader and firm issue

closed 2020-12-22 by Asizon