blheli-configurator/blheli-configurator #199: not getting ui after read like you should

v-dub-spaz, 2020-12-17:
after i connect it is blank .. then i click read and still blank .... it shows up top as read but no lower ui pop up
[uhg why no motor list.txt](https://github.com/blheli-configurator/blheli-configurator/files/5707065/uhg.why.no.motor.list.txt)
[uhg 2.txt](https://github.com/blheli-configurator/blheli-configurator/files/5707066/uhg.2.txt)

Asizon, 2020-12-17:
@v-dub-spaz  Can you attach a diff from your Betaflight setup? Maybe you dont correctly configure Esc Protocol since it comes disabled as default.

sadlersr, 2021-03-18:
Hello, I have a Holybro Tekko32 F3 4in1 ESC(45A) and when I connect, I too get a message that the read was successful, but the screen remains blank.  I am using iNav Configurator 2.6.1 to flash my flight controller and I have tried setting the ESC protocol to both Multishot and DShot600 without success.  Am I missing something obvious?  I've attached my 'diff' and BLHeli Configurator Log output.
[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/6160995/Log.txt)

[Diff.txt](https://github.com/blheli-configurator/blheli-configurator/files/6160979/Diff.txt)