blheli-configurator/blheli-configurator #210: esc wont read on jesc config after updating flight controller, says its not licnced

willantill, 2021-02-22:
[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/6022885/Log.txt)
[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/6022889/Log.txt)

Hi there 
I hope your well? and thanks for all the hard work hopefully I will get to enjoy it soon!

Order Details
--
Order ID: 23644Date Added: 22/02/2021Payment Method: PayPal | E-mail: williamantill@hotmail.co.ukTelephone: 07506792051IP Address: 86.18.92.76Order Status: Complete

I was watching joshua bradwells video on rpm filtering.

I have recently done a new build. Its a 5' running a mamba stack, f405 and f40 esc. it was running betaflifgt 4.1.2, I think it was. 

I flashed licnced the escs, flashed them with the firmwere for 4.1.2 betaflight and it was making a differnt chime. following bardwells video I checked for the error on the motors tab on beta flight and the had 

e:0.00% which was good. but checking the action in the CLI the gyro / pid max load was over 50%.

this was at 4k 4k dshot 300.
 I dropped it to dshot 150. max load was at 47% JB said that bad.
 
So i decieded to update to 4.2.  

since then I have 100% error on the motors tab on beta flight and the qwad wont arm, receiver is receiving etc. 
the motors wont spin on beta flight.

the esc wont read on jesc config as it says its not licnced. It was.

esc wont read on blheli config.

it wont make the full tune on battery conection.

i feel like its bricked.

Any help greatly apreciated as I have only flown this build for one pack!

best, Will

willantill, 2021-02-22:
I cant license the esc as all options are greyed out. the esc configurators both show empty when read. However the esc is not dead the first half of the new tune to show telemetry is playing but not the final few beeps. the motors also twitch.

willantill, 2021-02-22:
Iv just noticed when I read the 4 in 1 esc although no escs show up it says its protocol is 108. 

I have bought 4 more esc licences to see if it is possible to re licence the same esc but it is still greyed out after reading.

mathiasvr, 2021-02-22:
JESC only supports dshot 300/600 on H type ESCs. So if it works with dshot 300 that's probably what you should use.

willantill, 2021-02-22:
Thanks for a quick reply.

After the max lod was still high, I updated the FC to 4.2.8 and now the esc won’t read at all.

It says it’s unlicensed. I bought 4 more licences. After reading licence all is still greyed out. Configurator shows no escs but shows protocol 108.

Hope this helps

mathiasvr, 2021-02-22:
Are you using Jesc configurator? You might need to ask here instead https://github.com/jflight-public/jesc-configurator/releases

willantill, 2021-02-22:
Thanks for the reply dude! Yes I’m using the jacket config I’ll leave a post there thank you again