blheli-configurator/blheli-configurator #218: Error, layout not found. ESC position as yyyyyyyyy

Lean77, 2021-04-05:
Hi, i just flashed my ESC with BLheli 16.7, after spooling the motors up I realized that 2 motors were running differently. I checked BLheli and saw that their positions were registered as yyyyyy. I couldn't flash new firmware, are the ESCs bricked? 

Here is the error:
2021-04-05 @ 19:11:45 -- ESC 3: flashing failed - Error: LAYOUT found ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ, #M_H_50# expected at checkESCAndMCU (chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/build/blheli_configurator.js:767:47) at _fulfilled (chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/libraries/q.js:794:58) at chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/libraries/q.js:823:34 at Promise.promise.promiseDispatch (chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/libraries/q.js:756:17) at chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/libraries/q.js:564:48 at flush (chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/libraries/q.js:110:21) at processTicksAndRejections (internal/process/task_queues.js:75:11)

mathiasvr, 2021-04-05:
Hi, when flashing you need to check the `Ignore inappropriate MCU and Layout?` checkbox and carefully select the correct firmware layout, which should be `M_H_50` in your case.

Lean77, 2021-04-06:
yess! it worked, kinda panicked for a moment there, thx!

closed 2021-04-06 by Lean77