blheli-configurator/blheli-configurator #219: BlHeli wont read my ESC

hprompt, 2021-04-17:
Hi There,
I have a flywoo explorer LR4 analog with crossfire nano RX. I swapped my VTX out because I damaged the connector. I installed a TBS unity pro 5G8 HV  SE (MMCX) and it works fine. now I can't get it to arm so I connected it to BlHeli_S and it won't read the ESC setup.
betaflight is closed
connected USB to BlHeli configurator, connected battery, click read setup nothing after that. attached log.txt file 

[Log.txt](https://github.com/blheli-configurator/blheli-configurator/files/6329870/Log.txt)

mathiasvr, 2021-04-17:
Looks like your ESCs are not being read. Some things to check:
- Are the ESCs properly connected to the FC?
- Did you change betaflight settings / port mappings? 
- Do you get the correct startup beeps from the motors?

hprompt, 2021-04-17:
thanks for the quick repond
it was arming before I changed VTX
    I just changed vtx from goku vt625 to tbs unity pro 5G8 SE mmcx nothing else

Are the ESCs properly connected to the FC?
   
   -  I didn't change anything there   

   
   - Did you change betaflight settings / port mappings?

   
   - softserial 1  changed smartaudio from IRC tramp to TBS smartaudio   

   
   - Do you get the correct startup beeps from the motors?
   -    

    when I attach battery I get 3 beeps

mathiasvr, 2021-04-18:
The 3 initial beeps mean the ESCs are powered and connected to the motors.
You are missing the next two beeps, the first which is protocol detection.

Is the esc protocol e.g. dshot configured correctly in betaflight?

You may still want to confirm that the ESCs are properly connected to the FC.

hprompt, 2021-04-18:
YesDshot300 is configured that didn’t change