blheli-configurator/blheli-configurator #237: Cant flash 3 of 4 esc's. ESC: ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ, 16.6: ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ (includes error code and more)

bigjoncoop, 2022-03-06:
**ESC INFO:** Spedix IS30 / Q-H-20 / BLHeli_S
**BLHeli Configurator Version:** Standalone 1.2.0 & Chrome App & Speedybee (newest version)
**CAUSE:** Problem started after trying to flash ESC's firmware using the SpeedyBee built in ESC CONFIGURATOR 

**_SAME ISSUE TRYING TO FLASH / MAKE CHANGES IN BOTH BLHeli (windows) & ESC CONFIGURATOR (SpeedyBee app)_**
_(error seems to mention "chrome" a few times but same issue trying to use Speedybee for android also_)

**_PROBLEM:_** 
* 3 of 4 ESC's show " ESC: ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ, 16.6: ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ "
* After changing direction (Normal, Reversed, etc...) and selecting "Write Setup" it doesnt stick and reverts back
* After Selecting "Flash Firmware" the Type and Version don't show and say "Select ESC & Select Version"
* After selecting Appropriate Type and Version and selecting "Flash" it doesn't flash and gives error
**_(ERROR CODE BELOW AND LINK TO ENTIRE LOG FILE)_**

**_MORE INFO:_**
Recently I flashed my ESC's using the SpeedyBee Android app, trying to get startup tones working. I went to fly today and when I armed the quad it was apparent that some of the motors were turning the wrong direction. Which is strange since normally the ESC would retain the direction info after flashing( normal, reversed etc..) So I went to use the SpeedyBee app to check motor direction and change as necessary. When I noticed that 3 out of 4 ESC showed " ESC: ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ, 16.6: ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ "   I then noticed when I changed the direction and selected write setup that it would just revert back and not change. So when I got home i tried both Blheli_S chrome app and standalone the same 3 ESC's still showed " ESC: ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ, 16.6: ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ " and when trying to flash them I got a error code.

**LINK TO LOG FILE:**
* https://drive.google.com/file/d/1btpsGOua70l7sA8CrylB1ThBZdl-R6pK/view?usp=sharing
* https://pastebin.com/k1D4VDMr

**ERROR:**
`flashing failed - Error: LAYOUT found ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ, #Q_H_20# expected at checkESCAndMCU (chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/build/blheli_configurator.js:767:47) at _fulfilled (chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/libraries/q.js:794:58) at chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/libraries/q.js:823:34 at Promise.promise.promiseDispatch (chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/libraries/q.js:756:17) at chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/libraries/q.js:564:48 at flush (chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/libraries/q.js:110:21) at processTicksAndRejections (internal/process/task_queues.js:75:11)`

bigjoncoop, 2022-03-06:
*****UPDATE*****

I was able to fix the ESC's by connecting them to an Arduino and BLHeli_S suite and reflashing them.

elinho69, 2022-06-14:
como você conseguiu sou do Brasil e estou com o mesmo problema, mais em apenas um esc o esc 4